w3c-cg/web-nfc #452: Can I use Web-NFC to issue "commands" to NFC tags?

FernandoJAlves, 2019-11-26:
Hey!

Not sure if this is the best place to ask this, but I couldn't find a forum or slack for this project. Anyway, my scenario is the following: I'm trying to use Web-NFC to communicate with the e-passport (more specifically, implement the Basic Access Control protocol). To do so, the protocol requires the device (in this case, a smartphone) to request a challenge to the NFC chip (in the passport), and then the chip would respond with a random number.

I have been able to create a functional NDEFReader (and tested it with other NFC tags), and I believe it is working correctly (I can perform reads). It does not read anything with the passport, due to the way the e-passport works (this is the expected behaviour as far as I know).

My problem lies with issuing the challenge to the passport. As far as I can understand, Web-NFC only provides the "push" command to send information from the active device (smartphone) to the NFC tag (passport), correct? Well, I tried multiple combinations of NDEFPushOptions, but the result is always the same: the push stays "pending" after interacting with the passport (it doesn't even call the 'then' or 'catch' methods). The NDEFReader also does not receive anything upon interacting with the passport.

Is this type of interaction (issuing a command) not supported by Web-NFC? This is my first time using this API, so I may have overlooked something in the documentation. I have previously used a NodeJS application (pokusew/nfc-pcsc), that uses external NFC readers, to implement the Basic Access Control protocol. This application has a method "transmit", where I pass the command, and it gets sent to the passport correctly (and then rest of the protocol continues). If anyone has already used this application, I'm looking for a way to emulate the behaviour of "transmit" using Web-NFC.

Thanks!

P.S.: If there is another way to get in touch with the development team of Web-NFC (a forum, or Slack, or something similar), I would love to join in on the discussion.

zolkis, 2019-11-26:
At this time Web NFC does not support ISO 14443 / IsoDep / MIFARE yet, only NDEF.
Sorry for that, it's on the list. See #101 (quite old issue, granted).

FernandoJAlves, 2019-11-27:
Oh, that's unfortunate. I really wanted to create a web-based passport reader (to avoid using physical NFC readers).

I've read the comments on the issue you referred, just want to ask if you have an estimate of when Web-NFC might begin to support the feature I described? (the 'v2' that was mentioned in a few of the comments)

kenchris, 2019-11-27:
We need to ship v1 first :-)

Actually the reason who the reader is now called NDEFReader is so that we can create a TagTechReader or similar with the required API for dealing with lower level transport protocols

kenchris, 2019-11-27:
Some initial work done here:

https://github.com/w3c/web-nfc/issues/238

FernandoJAlves, 2019-11-29:
Ok, thank you for your quick responses

kenchris, 2020-04-22:
Actually the new TNEP protocol allows you to issue commands over NDEF:

https://github.com/w3c/web-nfc/issues/559

pawisoon, 2021-03-15:
Is there any update on the status of adding support for other ISO 14443 chips?

suttyct, 2021-05-26:
> Is there any update on the status of adding support for other ISO 14443 chips?

I would also like to follow up if there is any update on the support ISO 14443 chips / Biometric Passport reading

llermaly, 2021-06-09:
@FernandoJAlves Hello Fernando, did you have any luck on this? I'm facing the same sitatuation and I'm not sure on how to start.

Any guidance/code (with or without web NFC) would be appreciated.

thanks

FernandoJAlves, 2021-06-12:
> @FernandoJAlves Hello Fernando, did you have any luck on this? I'm facing the same sitatuation and I'm not sure on how to start.
> 
> Any guidance/code (with or without web NFC) would be appreciated.
> 
> thanks

Unfortunately I was unable to find any way to use Web-NFC to solve my problem. At the time I was working on a university project that involved using NFC, and after the answers I got in this thread I decided to use the physical NFC readers I mentioned in my post, instead of smartphones.

Since then I have not been keeping up with Web-NFC, so I don't know if it already supports this functionality.

Sorry for the lack of information, but I hope it somewhat helps

llermaly, 2021-06-13:
Using USB NFC reader is fine , could you provide any guidance ? Any example
code would de life saver

Thanks

FernandoJAlves, 2021-06-13:
For the USB NFC reader I'm almost sure we ended up using the pokusew/nfc-pcsc. That project still appears to be somewhat active, it had a release back in February, here's the link: https://github.com/pokusew/nfc-pcsc

I've looked for the repository of the university project I mentioned, but I no longer have access to it unfortunately. At the time, the project was proposed by a company, and they probably removed our access after my team was done

pawisoon, 2023-11-07:
Any update on this? Would love to be able to communicate with ICAO9303 chipped documents