w3c-cg/web-nfc #455: Implementation status page

beaufortfrancois, 2019-11-27:
It would be great if this repository would host an implementation status page so that web developers interested in Web NFC could quickly see which features are implemented in various browsers?

See https://github.com/w3c/picture-in-picture/blob/master/implementation-status.md and https://github.com/WebBluetoothCG/web-bluetooth/blob/master/implementation-status.md

@leonhsl Could you list in this issue the currently missing/implemented features we would use to populate a table?

beaufortfrancois, 2019-11-27:
For instance, "peer" NDEFPushTarget is not implemented in Chromium.

leonhsl, 2019-12-02:
In Chromium:
    
- Not support any operations on NFC peers, just NFC tags for now.
- Not support reading/writing for several well-known type records.

       1. Smart poster records
       2. Signature records
       3. Handover records
       4. Local type name (acting as embedded records)

- Not support writing embedded records, i.e.  `NDEFRecordInit#data` 
 does not accept `NDEFMessageInit` yet.
- Not support triggering `NDEFReader#onerror` to notify errors that happened while trying to read a nfc tag, e.g. the tag coming nearby is not ndef-compatible, or some other IO errors.
- Not support calling `NDEFReader#scan()` the second time on the reader object which is already in the middle of scan().

leonhsl, 2019-12-02:
I looked through spec and code and got the above list for now. May update later.

riju, 2019-12-02:
Thanks @beaufortfrancois and @leonhsl, I am sure many developers will try  WebNFC and pointing them to a single source of truth is a good idea.

We should definitely list features like (smart posters, signature records, etc ) which we might implement if we feel there's a critical mass of users. It would be too complicated to add "voting" I guess, but we will get the pulse more effectively.

@leonhsl Feel free to add short notes and last updated also in the [implementation.md like here](https://github.com/w3c/mediacapture-image/blob/master/implementation-status.md).

beaufortfrancois, 2019-12-23:
I've noted "overwrite" was not supported as well in Chromium for now.

leonhsl, 2020-01-24:
Update to https://github.com/w3c/web-nfc/issues/455#issuecomment-560290421.

Now, in Chromium:
    
- Not support any operations on NFC peers, just NFC tags for now.
- Not support reading/writing for several well-known type records.

       1. Smart poster records
       2. Signature records
       3. Handover records
       4. Local type name (acting as embedded records)

- Not support calling `NDEFReader#scan()` a second time on the reader object which is already in the middle of scan().

leonhsl, 2020-01-24:
Update to https://github.com/w3c/web-nfc/issues/455#issuecomment-568460963.

We already added the support for `NDEFWriteOptions#overwrite`.

beaufortfrancois, 2020-12-30:
I believe this issue and https://github.com/w3c/web-nfc/blob/gh-pages/EXPLAINER.md#changes-done-after-origin-trial-ot have done the job. I'll close it.

closed 2020-12-30 by beaufortfrancois