w3c-cg/web-nfc #456: Write explainer

beaufortfrancois, 2019-11-27:
As this spec will go through another TAG review, we should have an explainer as described at https://w3ctag.github.io/explainers

@kenchris @zolkis Is that something you can write?

zolkis, 2019-11-27:
We get a lot of questions about scope. 
It's currently NDEF, but there are a lot of use cases for other tech as well.

Thing is, the scope has changed from the original idea of exposing a web-specific projection of NFC.
Since the scope shifted to low level, we need to re-think the key use cases.
At the moment it's NDEF read and write. 

We need to include a plan about adding new features and extending the scope.

We can take the Explainer template section by section and make proposals here.

kenchris, 2019-11-27:
https://github.com/w3ctag/w3ctag.github.io/blob/master/explainers.md

beaufortfrancois, 2019-12-13:
Fixed with https://github.com/w3c/web-nfc/pull/469

closed 2019-12-13 by beaufortfrancois