w3c-cg/web-nfc #458: About definition of local type records

leonhsl, 2019-12-02:
By reading https://w3c.github.io/web-nfc/#dfn-local-type, I think we should make clear that `local type records` supported by our web nfc API are just some TNF-1 (well-known type) records, they have specific meanings to their containing records.

But, https://w3c.github.io/web-nfc/#mapping-local-type-to-ndef says they could be written as TNF-4 records, which I do not think is appropriate. TNF-4 records have their own regulations on type names, also they are not expected by the nfc forum spec to act as embedded records.

> If record's recordType is a standard local type to smart poster or a handover record, then set ndefRecord's TNF field to 1 (well-known type record), otherwise set ndefRecord's TNF field to 4 (external type record).

I think we should remove such "smartness". Let me know your concerns here.

leonhsl, 2019-12-02:
I mean, simply, if an embedded record's `NDEFRecordInit#recordType` is `abc`, then we should write it as a TNF-1 record, **NOT** a TNF-4 record.

If Web Devs really want this embedded record to be an TNF-4 record, they just need to provide `xyz.com:abc` as `NDEFRecordInit#recordType`.

zolkis, 2019-12-02:
That is right: mapping local type to NDEF is wrong. I will fix it now.

zolkis, 2019-12-02:
@leonhsl could you please check the commit above?
I had to make other changes, too, for instance icon records were not correctly handled.

leonhsl, 2019-12-03:
I added several comments in the commit, and I think @beaufortfrancois might also want to take a look there, especially from Web Devs' point of view.

commit e7f47e2b (2019-12-11): Fix #458, handling local types
closed 2019-12-11 by zolkis