w3c-cg/web-nfc #459: Tag with 'NFC-A' and 'ISO-DEP' technologies but not recognized as ndef-compatible

Honry, 2019-12-04:
I have a tag (see following image read from native nfc app), with 'NFC-A' and 'ISO-DEP' technologies and no NDEF message, which is type of NFC Forum Type 4, per spec https://w3c.github.io/web-nfc/#ndef-compatible-tag-types, my tag should be one of ndef compatible tag types, but which couldn't be read/write from both native nfc app or Chrome, when writing with native nfc app, it throws with "Non-ndef compatible" error. 

[image]

leonhsl, 2019-12-04:
By my understanding, this tag should just be ndef compatible because of the `NfcA` technology it carries.

Could anyone help clarify the phenomenon here?

kenchris, 2019-12-04:
Is is formatted as ndef?

Honry, 2019-12-04:
No, it could not be formatted as ndef.

zolkis, 2019-12-04:
AFAIK they MAY also expose NDEF, but not required to do so.

Honry, 2019-12-04:
So a tag is of NFC Forum Type 4 may be NDEF compatible, but a tag without NDEF technology (e.g. my tag with only "NFC-A" and "ISO-DEP" technologies) must be NDEF non-compatible, right?

kenchris, 2020-05-20:
as far as I understand all NFC tags should support ndef, but I might be wrong - or this might be an issue with the tag. But as this is outside our scope I am closing

closed 2020-05-20 by kenchris