w3c-cg/web-nfc #462: NDEFRecord.data should return null rather than undefined

Honry, 2019-12-12:
In following algorithms,
> 9.11.9.6 Parsing NDEF MIME type records
9.11.9.8 Parsing NDEF external type records
9.11.9.9 Parsing NDEF unknown type records

Spec says:
>  Let buffer be the byte sequence of ndefRecords's PAYLOAD field if that exists, or otherwise undefined.
Set record's data to buffer.

But per discussion in https://github.com/w3c/web-nfc/pull/441,  `null` is a more correct falsey value than `undefined` for an interface type.

Besides, setter of record's data is missing in empty record type's parser algorithm.

beaufortfrancois, 2019-12-12:
Thanks @Honry for raising this issue!
I'll open a PR to fix this soon.

beaufortfrancois, 2019-12-12:
I've just opened https://github.com/w3c/web-nfc/pull/463

PR #463 (2019-12-12): s/undefined/null/g for |data| when empty payload
FIX #462

***
<a href="https://pr-preview.s3.amazonaws.com/beaufortfrancois/web-nfc/pull/463.html" title="Last updated on Dec 12, 2019, 8:28 AM UTC (e89b33a)">Preview</a> | <a href="https://pr-preview.s3.amazonaws.com/w3c/web-nfc/463/27f97a9...beaufortfrancois:e89b33a.html" title="Last updated on Dec 12, 2019, 8:28 AM UTC (e89b33a)">Diff</a>
closed 2019-12-12 by beaufortfrancois