w3c-cg/web-nfc #465: Is an empty record type's id null or empty string?

Honry, 2019-12-12:
Per 9.11.9.1 Parsing records from bytes:

> If idLength > 0, let ndef's id be result of running UTF-8 decode on the next idLength (ID field) bytes, or else let ndef's id be the empty string.

It should be empty string.

But spec says 
> The value 0x00 (Empty) indicates that there is no type or payload associated with this NDEF record. When used, the TYPE_LENGTH, ID_LENGTH and PAYLOAD_LENGTH fields are zero and the TYPE, ID and PAYLOAD fields are omitted from the NDEF record.

@beaufortfrancois would prefer null instead of empty string.

@zolkis, @kenchris, what's your opinion?

zolkis, 2019-12-12:
Repeating here as well, when an NDEF field is not present, I would map it to `undefined`, or if that is not desired/possible, then `null`.

beaufortfrancois, 2019-12-24:
@Honry We've picked `null`. See https://github.com/w3c/web-nfc/pull/492

closed 2019-12-24 by beaufortfrancois