w3c-cg/web-nfc #470: Simplify creating external records

zolkis, 2019-12-13:
Following the discussion [here](https://github.com/w3c/web-nfc/pull/467#discussion_r357220894).

@beaufortfrancois: 
> Thinking about it again, I don't think Web NFC should be smart and use the effective domain of the origin as domain name in external type names when creating/reading external type records.
I would argue web developers should always set the full external type record. Think about staging environments, where `staging.example.com:smart-poster` would be written automatically if `:smart-poster` were to be allowed. It would not match when read in a prod environment.

So the proposal is to validate input for external type as follows:
- find the last occurrence of `:`; 
- the part after it should validate as an external type
- the part before it has to be a domain name that should be validated by the [valid domain](https://url.spec.whatwg.org/#valid-domain) steps from the URL Standard (thanks @kenchris for the pointer).

kenchris, 2019-12-13:
You might need to use this instead https://url.spec.whatwg.org/#host-parsing

kenchris, 2019-12-13:
Just doing this on the remaining string should probably be fine:

[image]

commit 7426f9f6 (2019-12-16): Fix #470: remove adding effective domain (#474)

* Fix #470: remove adding effective domain

  * Fix validate external type steps
  * Fix link to forbidden code point
  * Improve validating local type names
  * Fix usage of 'which'
  * Fix validate local type steps
closed 2019-12-16 by kenchris