w3c-cg/web-nfc #487: toRecords() should be nullable or not?

leonhsl, 2019-12-24:
I think `toRecords`() is better to be nullable.
It returning null indicates this `Sp/external` record's payload is just a plain buffer but not an NDEF message at all, and returning an empty sequence indicates the payload is an NDEF message that does not have any record.

Currently the spec https://w3c.github.io/web-nfc/#dfn-parse-records-from-bytes makes `toRecords`() return an empty sequence for both the cases mentioned above.

WDYT? The background is https://chromium-review.googlesource.com/c/chromium/src/+/1980586, we need to change either spec or impl to keep them consistent.
 @kenchris  @zolkis  @beaufortfrancois @riju

zolkis, 2019-12-24:
Makes sense to me. Once more it proves that implementation feedback is essential.

beaufortfrancois, 2019-12-24:
I'll update spec to make `toRecords` nullable.

closed 2019-12-24 by zolkis