w3c-cg/web-nfc #504: Add note about Web NFC conventions

beaufortfrancois, 2020-01-02:
@zolkis [said](https://github.com/w3c/web-nfc/pull/502#pullrequestreview-337667570):
> I suggest adding a Note or other clarification (at your discretion) that summarizes what new conventions Web NFC is making about type names vs the NFC Forum standardized type names.

beaufortfrancois, 2020-01-02:
Did I miss anything in this table?

### Differences between NFC Forum and Web NFC

| Record type name | NFC Forum (Type field) | Web NFC (recordType) |
| - | - | - |
Text record | `T` | `text` 
URL record | `U` | `url` 
Smart poster record | `Sp` | `smart-poster` 
Local record (example) | `act` | `:act`
Empty record | | `empty`

zolkis, 2020-01-02:
Nice, thanks! We should also add `t` vs `:t` for smart poster local type "type", and `s` vs `:s` for local type "size" record. Title could perhaps be "Mapping between NFC Forum standard types and Web NFC type names" or similar?

kenchris, 2020-01-02:
This is only mapping of "well known types" and their encoding. NFC has additional types, like absolute url, external etc

zolkis, 2020-01-02:
Right, actually we could embed/merge that table with the tables in [Data mapping](https://w3c.github.io/web-nfc/#data-mapping).

beaufortfrancois, 2020-01-03:
@zolkis I've followed your advice. See https://github.com/w3c/web-nfc/pull/506

closed 2020-01-03 by beaufortfrancois