w3c-cg/web-nfc #505: How to read/parse local types

beaufortfrancois, 2020-01-03:
The "parse an NDEF record" used when reading "sub records" doesn't say explicitly how to deal with local types such as ":act", ":s", etc. It only deals with "T", "U", and "Sp".

@zolkis Is that intentional ?

[image]

zolkis, 2020-01-03:
Yes, since those are (NFC Forum standardized) local types to `Sp`.

beaufortfrancois, 2020-01-03:
Where would be the best place then to mention that a colon needs to be inserted at the beginning of recordType when record is of a local type?

zolkis, 2020-01-03:
With the new prefixing convention we have no other way than to complete the [parse Sp](https://w3c.github.io/web-nfc/#dfn-parse-an-ndef-smart-poster-record) algorithm.

beaufortfrancois, 2020-01-03:
But this algorithm uses back the "parse an NDEF record" algorithm.

zolkis, 2020-01-03:
Right. Completing the "parse Sp" algorithm will also need passing the context to the "parse record from bytes".

Currently "parse Sp" says

> 4. Let buffer be the byte sequence of ndefRecords's PAYLOAD field.
> 5. Set record's data to buffer. 

Then the app can call `toRecords()` on that record ('s data) which invokes [parse records from bytes](https://w3c.github.io/web-nfc/#dfn-parse-records-from-bytes). 

That has been fine when prefixing was not there, but now we need to pass the context to the invocation of [parse records from bytes](https://w3c.github.io/web-nfc/#dfn-parse-records-from-bytes) from within `toRecords()` and handle local types' parsing there. We cannot just pass local types any more, we need to translate them. 

I already feel nostalgic about the IMHO simpler `isLocal` :), but we should have passed context info to the parse algorithm anyway, so this has always been a miss, only uncovered now.

zolkis, 2020-01-03:
This issue also tracks the Sp part mentioned in #408.

leonhsl, 2020-01-10:
2 observations relating to local types.

1. https://w3c.github.io/web-nfc/#example-12, "act" is not a valid record type in our WebNFC world, should be ":act".

>    } else if (record.recordType == "act") {

2. https://w3c.github.io/web-nfc/#dfn-convert-ndefrecord-data-bytes, local type is missing. It should also support `toRecords()`.

leonhsl, 2020-01-13:
Currently https://w3c.github.io/web-nfc/#dfn-validate-local-type just validates like

> If localTypeName does not start with a lowercase character or a number, return false.

but how about the rest of the type name?

I feel we better have the same validation rule as the type part of external type name, but unsure how the nfc forum spec regulates it.
> If type contains code points that are not ASCII alphanumeric, or U+0024 ($), U+0027 ('), U+0028 LEFT PARENTHESIS ((), U+0029 RIGHT PARENTHESIS ()), U+002A (*), U+002B (+), U+002C (,), U+002D (-), U+002E (.), U+003B (;), U+003D (=), U+0040 (@), U+005F (_), return false.

leonhsl, 2020-01-13:
And I'd like to put all concerns about local type in this issue, another concern: https://github.com/w3c/web-nfc/issues/511#issuecomment-571946449

zolkis, 2020-01-13:
For local types the NFC Forum doesn't ask for more that starting with lowercase char or number.
There is no restriction for the rest. It's left up to the application.

If we see Web NFC a kind of application, we can define some extra rules if we can justify them. We should stay flexible for reading (for interoperability) but we can be stricter on writing, if needed. So far I haven't seen the reason to do that.

The external type names are more strictly regulated by NFC Forum.

leonhsl, 2020-01-13:
Happy to know that.

And agree that we read from the nfc tag flexibly (for interoperability), our impl is already in this direction :) by exposing to JS what has been read as is.

PR #536 (2020-03-11): Improve parsing smart poster: add missing steps
Changed the parse record steps to also receive a context.
Fixes #505 .
Likely improvements are needed on formulating the steps, @beaufortfrancois any suggestions?
@leonhsl PTAL

***

### :boom: Error: 500 Internal Server Error :boom: ###

[PR Preview](https://github.com/tobie/pr-preview#pr-preview) failed to build. _(Last tried on Mar 10, 2020, 3:39 PM UTC)_.

<details>
<summary>More</summary>

PR Preview relies on a number of web services to run. There seems to be an issue with the following one:

:rotating_light: [Spec Generator](https://www.w3 [... cut]
closed 2020-03-11 by zolkis