w3c-cg/web-nfc #521: NDEFRecordInit#encoding should be only for BufferSource data source

leonhsl, 2020-01-08:
https://w3c.github.io/web-nfc/#mapping-string-to-ndef needs to be refined.

If the data source is a string, we always get its `utf-8` bytes and write as the payload, i.e. `encoding` is forcibly set to "utf-8".

`NDEFRecordInit#encoding` is only used for WebDevs to declare how the `BufferSource` data is encoded.

beaufortfrancois, 2020-01-08:
You're right. I'll fix this.

closed 2020-01-08 by beaufortfrancois

leonhsl, 2020-01-10:
Should we throw an error if `NDEFRecordInit#encoding` is neither 'utf-8' nor `undefined` for a DOMString data source?

I feel necessary as I always prefer tight checks, thus users passing 'utf-16' for a DOMString will not get a wrong understanding that the DOMString is successfully encoded as 'utf-16' and written into the nfc tag finally.

The context is https://chromium-review.googlesource.com/c/chromium/src/+/1991972/1/third_party/blink/renderer/modules/nfc/ndef_record.cc#153

PR #531 (2020-01-10): encoding should be only for BufferSource data source
As raised by @leonhsl, we should throw an error if `NDEFRecordInit#encoding` is neither 'utf-8' nor `undefined` for a DOMString data source.

FIX #521

***
<a href="https://pr-preview.s3.amazonaws.com/beaufortfrancois/web-nfc/pull/531.html" title="Last updated on Jan 10, 2020, 10:31 AM UTC (bb9b35d)">Preview</a> | <a href="https://pr-preview.s3.amazonaws.com/w3c/web-nfc/531/1f90f6f...beaufortfrancois:bb9b35d.html" title="Last updated on Jan 10, 2020, 10:31 AM UTC (bb9b35d)">Diff</a>