w3c-cg/web-nfc #523: Handling pushed msg exceeds tag's size and how is the msg size calculated

Honry, 2020-01-09:
Spec should handles when pushed message exceeds the NFC tag's memory size, and may worth to explain how is the pushed message size calculated or maybe just leave it to implementer with a simple Note.

leonhsl, 2020-02-04:
I think we need to figure out this from the 2 aspects:
1. Is it necessary to put a limit on the size of the NDEF message pushed via `NDEFWriter#write()`?
2. Is it necessary to put a limit on the size of the NDEF message that can be dispatched to `NDEFReader#onreading`?

The old (current) WebNFC impl in Chromium imposes a limitation of 32K bytes on both conditions above, but does not implement well in respect of the algorithm for calculating the size of an NDEF message.

WDYT?  @zolkis @beaufortfrancois @riju @reillyeon @kenchris

zolkis, 2020-02-04:
> Spec should handles when pushed message exceeds the NFC tag's memory size

Then the app will get an error.
I don't think the spec _have_ to handle this, maybe nice-to-have, but then impl needs to figure out the characteristics of the tag and the underlying NFC implementation and then do the calculation. It is also difficult to fully test it, especially with automated tests and I am not sure that should fall on implementations. I think this is applications' responsibility.

leonhsl, 2020-02-04:
Yep this is also what I'm feeling.

So I prefer to remove the 32k bytes limitation from the current Chrome implementation, given that it does not make much sense and the implementation of size calculation is quite buggy.

closed 2020-05-20 by kenchris