w3c-cg/web-nfc #529: P2P support

kenchris, 2020-01-09:
Android is deprecating SNEP and thus the current peer to peer support

> The Simple NDEF Exchange Protocol (SNEP) is an application-level protocol suitable for sending or retrieving of application data units, in the form of NDEF messages between two NFC Forum Devices operating in NFC Forum Peer Mode

Answer from Android:
> However most of the functionality can be provided using the card emulation interface. NFC Forum type 4 tag specification describes a mechanism for NDEF messages exchange

The current P2P support is a one time operation - when read, its done, where as HCE (Host Card Emulation) most probably isn't.

It might be possible to implement the existing peer support using HCE, but actually being able to not cancel on read might also be useful and affect API design.

Thus maybe it makes more sense to add a NDEFTagEmulation object or similar, as a specialization of HCE

kenchris, 2020-01-09:
https://stackoverflow.com/questions/29122848/ndef-message-with-hce-android

kenchris, 2020-01-09:
https://github.com/justinribeiro/android-hostcardemulation-sample

kenchris, 2020-01-09:
https://nfc-forum.org/nfc-forum-statement-regarding-host-card-emulation-hce/

kenchris, 2020-01-09:
Android supports HCE

* Support for NFC-B (ISO/IEC 14443-4 Type B) technology is optional
* Support for NFC-A (ISO/IEC 14443-3 Type A) technology is mandatory
* Support for NFC-Forum ISO-DEP specification (based on ISO/IEC 14443-4) is mandatory and must be emulated on top of NFC-A. 
  * Hence, the emulated tag is a very similar to NFC Forum Type 4, though NFC-B support is optional.
* Process Application Protocol Data Units (APDUs) (ISO/IEC 7816-4 - smart card) 
  * This comes from ISO-DEP which uses ISO/IEC 7816-4 APDU commands
  * This is what allows emulating NDEF

[image]

kenchris, 2020-01-09:
I am trying to understand https://developer.apple.com/documentation/corenfc/nfciso7816tag

I guess that is how it can read HCE emulated tags, but it this just the same as reading NFC Forum Type 4 tags or can it read additional tags?

cyberphone, 2020-01-09:
There is a bunch of mobile payment systems out there including AliPay which do not build on low level card protocols like EMV (which is security-wise unsuited for the Web).  The systems mentioned are currently stuck with QR both at the Desktop/Web and at the POS terminal.

justinribeiro, 2020-01-09:
> I am trying to understand https://developer.apple.com/documentation/corenfc/nfciso7816tag
> I guess that is how it can read HCE emulated tags, but it this just the same as reading NFC Forum Type 4 tags or can it read additional tags?

I haven't yet tried to write an example with that on iOS, but based on my reading of the docs it reads to me like that would be specific to Type4. I'm on the road for the next couple of weeks, but I'll have to test that theory when I get back to office.

kenchris, 2020-01-09:
@justinribeiro what is the license of your sample ndef emulation code? Maybe we could base our on that if you are willing to relicense to the Chromium Project

justinribeiro, 2020-01-10:
We can relicense that sample to the Chromium Project, I'm cool with that @kenchris.

asoltys, 2021-01-23:
Hi, I'm eagerly awaiting some kind of P2P NFC support to land in Chromium so I can use it in a PWA bitcoin wallet app I've made. I made a [short video](https://www.youtube.com/watch?v=IVNhW63GF9U) demoing the non-P2P web NFC functionality a few months ago. Is #567 looking like the best hope at this point?

eklem, 2022-01-11:
A nice use-case for peer-to-peer support could be transferring one-time pads for encryption/decryption.

kenchris, 2022-06-29:
This is what TNEP is for: https://github.com/w3c/web-nfc/issues/563

closed 2022-06-29 by kenchris