w3c-cg/web-nfc #537: Security and Privacy: NDEF scope not considered adequately

micolous, 2020-01-26:
Per the [Security and Privacy Questionnaire](https://github.com/w3c/web-nfc/blob/gh-pages/security-privacy-questionnaire.md#23-how-does-this-specification-deal-with-personal-information-or-personally-identifiable-information-or-information-derived-thereof), it argues that the scope is limited because this API only processes NDEF records.

However, as mobile NFC access has been historically highly limited on non-Android devices, NDEF is a _de facto_ lowest common denominator for NFC access.  As a result, it is not possible to anticipate how NDEF records are used, or in what ways it has been "repurposed".

For example, [Yubikey has an "NDEF compatibility mode"](https://support.yubico.com/support/solutions/articles/15000006432-understanding-the-ndef-interface-on-nfc-enabled-yubikeys), where the device will pass back OTPs in text form (to be pasted into an app/browser on the phone).

The security and privacy considerations should not presume `NDEF is a post-it note`, and consider things like tags returning dynamic data over NDEF, or data that could be passed through NDEF records.

zolkis, 2020-01-27:
It is not clear what is the threat in the Yubikey case. NDEF fallback is quite often used. The content owner can decide what information to put there. 

Related, it is not clear why and in what ways implementations or browser vendors are liable to _anticipate_ how NDEF content is used (by an app, or other apps, I assume).

The "post-it note" metaphor was used for explaining that an NDEF tag can be read by anyone.
That is slightly different from peer to peer (discontinued) which is dynamic and targeted (in a similar fashion you describe plain NDEF can also be used), but the Security section did cover P2P as well. 

Please bear with me: in order to improve things we need more specific threat descriptions and eventually mitigation proposals, too.

beaufortfrancois, 2020-03-30:
For info, a blocklist containing vulnerable YubiKeys has been added to the spec. See https://github.com/w3c/web-nfc/issues/550 and https://w3c.github.io/web-nfc/#blocklist

closed 2020-05-20 by kenchris

micolous, 2021-12-27:
Closing the loop (on a big delay, sorry!): it looks like someone else explained it for the Yubikey case in #543 (NDEF fallback data contains an OTP).

However, this is a _single_ device out of many, and I don't think a blocklist based on historical bytes is the answer:

* [there is no specified update frequency for the blocklist](https://w3c.github.io/web-nfc/#blocklist), nor an expiry mechanism.
* cards of same model have the same ATR (and thus historical bytes), _even when used for different purposes_ (eg: storage cards like MIFARE Ultralight and NTAG).
* the same model of card has may have a different ATR (and thus historical bytes) on different readers. Ludovic Rousseau (author of `pcsclite`) maintains [a public list of ATRs](http://ludovic.rousseau.free.fr/softwares/pcsc-tools/smartcard_list.txt) where you can see this issue.
* [some cards have no historical bytes at all][ad], so are impossible to add to an allow or block list.
* filtering based on content is similarly fraught with technical difficulties.

[ad]: https://developer.android.com/reference/android/nfc/tech/IsoDep#getHistoricalBytes()

[A better answer would be to give a way for tags to explicitly declare compatibility](https://github.com/mozilla/standards-positions/issues/238#issuecomment-610691221). This could be done with an NDEF record extension, but is something you'd need to propose to the NFC forum.

mseddon, 2022-05-05:
> Closing the loop (on a big delay, sorry!): it looks like someone else explained it for the Yubikey case in #543 (NDEF fallback data contains an OTP).
> 
> However, this is a _single_ device out of many, and I don't think a blocklist based on historical bytes is the answer:
> 
> * [there is no specified update frequency for the blocklist](https://w3c.github.io/web-nfc/#blocklist), nor an expiry mechanism.
> * cards of same model have the same ATR (and thus historical bytes), _even when used for different purposes_ (eg: storage cards like MIFARE Ultralight and NTAG).
> * the same model of card has may have a different ATR (and thus historical bytes) on different readers. Ludovic Rousseau (author of `pcsclite`) maintains [a public list of ATRs](http://ludovic.rousseau.free.fr/softwares/pcsc-tools/smartcard_list.txt) where you can see this issue.
> * [some cards have no historical bytes at all](https://developer.android.com/reference/android/nfc/tech/IsoDep#getHistoricalBytes()), so are impossible to add to an allow or block list.
> * filtering based on content is similarly fraught with technical difficulties.
> 
> [A better answer would be to give a way for tags to explicitly declare compatibility](https://github.com/mozilla/standards-positions/issues/238#issuecomment-610691221). This could be done with an NDEF record extension, but is something you'd need to propose to the NFC forum.

It's unclear how this would avoid tags which suck without realising it claiming they don't however. I'm honestly not so far convinced with the arguments presented so far.

anssiko, 2022-05-06:
In a Web NFC Community Group chair capacity I want to say the group in general prefers to see concrete, specific issue descriptions over abstract descriptions. An issue that can be reproduced in real life will be more helpful as it allows us to apply engineering methodologies to improve things. I thank everyone participating in this issue for their contributions that work toward that goal.

This group has taken and continues to take a data-driven and evidence-based approach to security and privacy and this is complemented by each browser vendors' own reviews conducted separately. I should remind that browser vendors' ship decisions are out of scope from the specification discussion in this repo and such feedback is more impactful if directed at respective browser vendors' feedback channels. Also, real-world deployments of this technology, even if smaller scale, are a good way to demonstrate the benefit to implementers.