w3c-cg/web-nfc #538: Definitions of NDEFMessageInit and NDEFRecordInit violates Web IDL spec

peria, 2020-01-28:
Two IDL dictionaries 'NDEFMessageInit' and 'NDEFRecordInit' are defined as following;
```
dictionary   {
  required sequence<NDEFRecordInit> records;
};
dictionary NDEFRecordInit {
  ...(snip)...
  NDEFRecordDataSource data;
};
typedef (DOMString or BufferSource or NDEFMessageInit) NDEFRecordDataSource;
```

This reference cycle violates the description "The type of a dictionary member must not include the dictionary it appears on." in [Web IDL spec](https://heycam.github.io/webidl/#idl-dictionaries)

reillyeon, 2020-01-28:
I am curious about the reasoning behind the WebIDL specification intentionally banning recursive types. As a workaround we could define `NDEFRecordDataSource` as follows which should only trivially change the code developers need to write when creating embedded records,

```
typedef (DOMString or BufferSource or NDEFMessage) NDEFRecordDataSource;
```

zolkis, 2020-01-28:
Right, good catch. How are we supposed to define recursive structures with Web IDL then?
Use interface with 3 constructors instead of dictionary for NDEFRecord?

peria, 2020-01-29:
I think this issue is not urgent nor serious while most browsers are working well. :)
And I think it is a practical option to update Web IDL spec not to ban the cyclic references in dictionaries.

rakuco, 2021-09-17:
This ended up being filed again as #620 after it bit us in the Chromium implementation :/

reillyeon, 2021-09-17:
Shall we mark this as a duplicate?

rakuco, 2021-09-17:
Yeah, I think that makes sense, as all the recent discussion's tracked there.