w3c-cg/web-nfc #543: YubiKey NDEF analysis

kenchris, 2020-02-11:
NDEF can be considered post it notes. Things stored are as good at public information, so if you want to make sure that your content is only readable by certain parties, then make sure it is encrypted.

That brings us to security keys like YubiKey that can expose things over NDEF: https://support.yubico.com/support/solutions/articles/15000006432-understanding-the-ndef-interface-on-nfc-enabled-yubikeys

- One time passwords
- Static passwords
- OATH-HOTP

One time passwords are safer than static passwords as they are time restricted, often down to a few seconds. Static passwords are not safe to generally make public (like adding them to an NDEF record or a post it note) which is why YubiKey also makes it clear that they should part of your password (thus basically a less safe second factor like authentication):

https://support.yubico.com/support/solutions/articles/15000006480-understanding-core-static-password-features

Thus the NDEF support in YubiKey is only intended to compliment an existing password, and thus add additional security. We could argue that allowing static passwords is a really bad idea as NDEF is not considered safe, but it has been Yubico choice to do so and they warn against it and it is an advanced setting that manually needs to be enabled.

kenchris, 2020-02-11:
In traditional OTP solutions like RSA SecurID dongles, it is possible for users to have apps open the camera and scan the codes, but it requires the user to bring the camera within the field of view of the dongle - same as with NFC, you would need to place your unlocked phone within reach (5-10 cm, most likely 5cm) of the YubiKey in order to be able to read it.

You might think that that could happen with both in the pocket at the same time, but Web NFC adds further requirements:

* Unlocked phone
* Focused web app with NFC permission

Additionally, it could be possible to use the proximity sensor to disable NFC reads when something is close, like when in the pocked or when laying on the table face down with keys on top.

kenchris, 2020-02-11:
The Yubico static passwords and Yubico OTP can be used alone (OTP can even include username).

It is not really a safe practice. In many ways it feels like a physical key, as it can be copied and then abused, but in contract to physical keys, copying is quite quick if shared to the wrong person/service.

The time restriction of OTP makes that somewhat less of an issue, but at the end of the day, Yubikey leaves it up to the implementer (the user of the yubikey feature for their system) whether to make the compromise to use this less safe option in order to save money on the implementation side.

kenchris, 2020-02-11:
[image]

https://support.yubico.com/support/solutions/articles/15000014219-yubikey-5-series-technical-manual

kenchris, 2020-02-11:
[image]

https://www.yubico.com/wp-content/uploads/2015/03/YubiKeyManual_v3.4.pdf

kenchris, 2020-02-11:
NDEF programming on NEO:

[image]

https://www.yubico.com/wp-content/uploads/2016/03/YubiKeyPersonalizationToolUserGuideMarch2016.pdf

kenchris, 2020-02-11:
Another question that might come up: Do we need to worry about WebAuthn+FIDO2.

Short answer: No, it is based on APDU:

https://fidoalliance.org/specs/fido-v2.0-ps-20190130/fido-client-to-authenticator-protocol-v2.0-ps-20190130.html#nfc

zolkis, 2020-02-13:
Is there some mitigation we should include in the Security section?
AFAICT the current ones cover it, eventually we can make a Note explaining this specific case, since the wide publicity.

Web NFC would serve Yubikey data to the page, which can choose to share it forward. 
IIUC the criticism is that Web NFC doesn't enforce a policy that would prevent that.

Note that the group has investigated an origin policy for Web NFC, but that cannot be enforced because of the NFC standard (unless only encrypted content is used for Web NFC, with a special format - but then it becomes an application that is incompatible with the majority of NFC use cases).
However, the threats of using Web NFC are not more than using NFC per se, as 3rd parties (pages) cannot access Web NFC data of the page using Web NFC, unless the page shares it (in which case the use case is _intended_ by the NFC data provider, otherwise it would have encrypted the data).

Restricting further sharing is up to the permissions needed for the mechanisms for further sharing of the data obtained from Web NFC. This is as much the problem of the whole web platform as it is of Web NFC, so I don't see the reason to not offer the API for this particular reason (while a fetch API and others with similar problems exist where once obtained, the data could be arbitrarily shared further, without tracking the origin in continuation).

dbaron, 2020-02-13:
From the [initial comment](https://github.com/w3c/web-nfc/issues/543#issue-563213694):

> it is an advanced setting that manually needs to be enabled.

I've been looking at the various documentation links on `yubico.com` and I haven't found anything saying this -- where do you see it?  Everything I've read made it sound like the `NDEF` interface is enabled by default but there's advanced configuration that allows switching which of the two programmable slots is sent via `NDEF`.  In particular, it seems like the default configuration is that [slot 1 is a Yubico OTP credential](https://support.yubico.com/support/solutions/articles/15000014219-yubikey-5-series-technical-manual#OTPvdv167) and that [it is also the default for what is sent via `NDEF`](https://support.yubico.com/support/solutions/articles/15000010241-setting-the-ndef-slot-for-nfc-usage).

kenchris, 2020-02-14:
> it is an advanced setting that manually needs to be enabled.

What was behind an advanced setting was the "static password" feature. 

[image]

https://support.yubico.com/support/solutions/articles/15000006432-understanding-the-ndef-interface-on-nfc-enabled-yubikeys

flaki, 2020-03-09:
I'm not intimately familiar with the current state of the spec or the NFC driver exposed information, which is why I'm curious: is there a way for the browser (not the website, as that would be a fingerprinting risk) to know what device it is currently communicating with? Something akin to VID/PIDs of USB or similar, that would identify the vendor/device make.

I could imagine that certain implementations could choose to harden this particular feature/use case by showing an extra warning or a different permission dialog for accessing security-sensitive information (in particular, accessing devices known to expose OTP-s and passwords via the NDEF interface).

zolkis, 2020-03-09:
A blacklisting mechanism could be implemented, akin to Web Bluetooth [blocklist](https://webbluetoothcg.github.io/web-bluetooth/#the-gatt-blocklist).
However, won't be foolproof as the UID's can be false/duplicated on fakes so it can result in DoS attack on certain vendors. 

The more useful thing browser implementations could do is indeed to include information in the permission prompt about the tag tech and content type(s) before that data is handed over to the web page.

beaufortfrancois, 2020-03-10:
There is a way to identify YubiKeys based on their [historical bytes](https://developer.android.com/reference/android/nfc/tech/IsoDep#getHistoricalBytes())
See my WIP patch at https://chromium-review.googlesource.com/c/chromium/src/+/2078550

beaufortfrancois, 2020-03-30:
For info, we've added a blocklist to the spec at https://github.com/w3c/web-nfc/issues/550

closed 2020-05-20 by kenchris