w3c-cg/web-nfc #547: Other records accepted by Web NFC in smart poster

zolkis, 2020-03-06:
Captured from the discussion in #358 .
Questions:

1. Should be other records allowed that are not explicitly specified in NFC Forum Smart Poster RTD?
Yes, other records are allowed but applications MAY ignore it.
Web NFC implementation is an application from NFC Forum point of view, so it can define its own policies on what to accept.

2. What other records are allowed?
Consensus:
- MIME records are allowed
- only one URI, type size and action record is allowed
- the NFC Forum doesn't forbid other record types but there is a dispute whether to allow empty, absolute URL, external, local (other than Sp local), unknown records.

zolkis, 2020-03-06:
In my opinion,
- Absolute URL records have a URL so must not be used in SP.
- Empty records are anyway ignored on reads and seem to have no use case for writing.
- Unknown records don't seem to have a use case at least for creation. Reads could support it.
- External and local types should be OK to accept.

kenchris, 2020-03-06:
Just allow it all.. it might not make sense, but no reason to enforce arbitrary restrictions

zolkis, 2020-03-06:
OK - except absolute URL.
There MUST be one URL in the SP and that has to be a URI record.
Also, impls MUST remove URI/abs URI records from messages which contains a SP record.

kenchris, 2020-05-20:
Can this be closed @zolkis ?

zolkis, 2020-05-20:
Yes.

closed 2020-05-20 by zolkis