w3c-cg/web-nfc #549: Error event handling should specific the error code

Honry, 2020-03-11:
In [NFC reading algorithm](https://w3c.github.io/web-nfc/#dfn-nfc-reading-algorithm):
> If the NFC tag in proximity range does not expose NDEF technology for reading or formatting, run the following sub-steps:
> 1. For each NDEFReader instance reader in the activated reader objects, run the following sub-steps:
>    1. Fire an event named "error" at reader.
> 2. Abort these steps.

Spec doesn't explicitly specify the error code when firing 'error' event.

beaufortfrancois, 2020-03-11:
I don't think we need the error code in this case as the `error` event is a custom error event, not an DomException error. Please let me know why you think we need one.

Honry, 2020-03-11:
ECMAScript error objects (e.g., Error) and DOMException are encouraged to be used in Web standard, see https://w3ctag.github.io/design-principles/#error-types

beaufortfrancois, 2020-03-11:
Do you have an example of a Web API that uses an error event with an error object?

Honry, 2020-03-11:
Moreover, we may need to handle more errors, e.g. when user denies the permission during nfc reading.

beaufortfrancois, 2020-03-11:
I've found https://www.w3.org/TR/generic-sensor/#sensor-onerror

beaufortfrancois, 2020-03-11:
See why error event doesn't have error anymore: https://github.com/w3c/web-nfc/issues/391

Honry, 2020-03-11:
But per @reillyeon's comment https://github.com/w3c/web-nfc/issues/391#issuecomment-543843645, this is the same situation with Generic Sensor API that we have to handle errors during reading.

beaufortfrancois, 2020-03-11:
See discussion at https://github.com/w3c/web-nfc/pull/439#pullrequestreview-313554303

zolkis, 2020-03-11:
In the [scan()](https://w3c.github.io/web-nfc/#the-scan-method) algorithm we reject the promise with a DOMException when the NFC adapter is not accessible (many reasons given).

The reading algorithm is invoked by the scan() algorithm and will use a simple error event for any read error. An app doesn't need to know anything else, since the only way to recover is that the user retries the touch, therefore the app just has to show there was an error. 

We don't have a use case for exposing HW specific error codes to web pages and in general we better avoid it in order to minimize fingerprinting surface.

Honry, 2020-03-11:
@beaufortfrancois, @zolkis, many thanks for your explanation! Now I understand why we don't explicitly specify the error. BTW, shall we handle if user deny permission during nfc reading?

zolkis, 2020-03-11:
Obtaining permission is handled by scan(). At the time of reading, the permission is either granted or rejected in the interaction flow. After that, the user should only have the option to either touch a tag or cancel the operation.

Or do you mean that depending on implementation, the permission dialog might be persistent over time during the whole session and a user could deny permission even while making a gesture to touch a tag?

That would be bad UI and it would be quite schizophrenic from a user. What I expect to happen is a race condition in the underlying platform + HW. At that point the read might be successful, but the implementation could still decide to not report the NDEF message (and eventually fire an error event). For this eventuality we could include an extra step in the reading algorithm.

However, this scenario seems invalid to me. As said above, in the Web NFC interaction flow the permission is handled by scan(), and if the user changes opinion, then has the options to not touch a tag and/or cancel the operation.

Honry, 2020-03-11:
I was thinking of a scenario from testing perspective, Chrome allows user to deny the permission from "Site Settings" menu, I know it‘s rare that end user happen to do such operation. Just curious if spec need to take care of this scenario. :)

beaufortfrancois, 2020-03-11:
I use permissions API to track this changes:

```js
navigator.permissions.query({ name: "nfc" }).then(permissionStatus => {
  console.log(`NFC user permission: ${permissionStatus.state}`);
  permissionStatus.onchange = _ => {
    console.log(`NFC user permission changed: ${permissionStatus.state}`);
  };
});
```

closed 2020-03-12 by Honry