w3c-cg/web-nfc #550: Create a list of device IDs we will explicitly block in Web NFC

beaufortfrancois, 2020-03-12:
We should create a list, similar to the ones in the WebUSB and Web Bluetooth repos, of device IDs we are explicitly blocking. The first ones to be included will be YubiKey devices as raised in https://github.com/w3c/web-nfc/issues/543

Here's the WebUSB blocklist: https://github.com/WICG/webusb/blob/master/blocklist.txt

For info, here's the chromium CL that introduces the blocklist: https://chromium-review.googlesource.com/c/chromium/src/+/2078550

beaufortfrancois, 2020-03-12:
It could look like as something like this

```
# Historical bytes

0x80, 0x73, 0xc0, 0x21, 0xc0, 0x57, 0x59, 0x75, 0x62, 0x69, 0x4b, 0x65, 0x79 #YubiKey 5 Series
0x59, 0x75, 0x62, 0x69, 0x6b, 0x65, 0x79, 0x4e, 0x45, 0x4f, 0x72, 0x33 # YubiKey NEO 
```

beaufortfrancois, 2020-03-12:
cc @reillyeon

reillyeon, 2020-03-12:
We should make sure that the fields being matched are well-defined in terms of the NFC specifications. I'm not sure if "historical bytes" is a term from NFC or the Android API.

closed 2020-03-30 by beaufortfrancois

miguelUS, 2022-02-24:
So are YubiKeys NFC blocked from Web NFC? 
Can a developer remove the blocking if he/she chooses to?

beaufortfrancois, 2022-02-24:
> So are YubiKeys NFC blocked from Web NFC? 
Yes. See https://source.chromium.org/chromium/chromium/src/+/main:services/device/nfc/android/java/src/org/chromium/device/nfc/NfcBlocklist.java

> Can a developer remove the blocking if he/she chooses to?
Nope. See https://source.chromium.org/chromium/chromium/src/+/main:services/device/nfc/android/java/src/org/chromium/device/nfc/NfcTagHandler.java;l=35;drc=d3a5b84f152932d25c0d42d345567c6c40bd768e

What is your use case @miguelUS?

beaufortfrancois, 2022-02-24:
Yes. This Blocklist is there to block this kind of attacks. See discussion at https://github.com/w3c/web-nfc/issues/543#issue-563213694