w3c-cg/web-nfc #556: Thank You... Seriously

lyellick-dev, 2020-04-03:
I personally have been really looking forward to a way to interact with NFC tags within a browser. I have tried android/xamarin frameworks for NFC interaction and just couldn't get past the learning curve. Wasn't until a few weeks ago I stumbled upon the documentation. Was so excited I took a personal day at work to play with this.

It feels really cool to be apart of a cutting edge feature when it seems like everything has been done or standard libraries have been built/recognized.

How i'm using Web NFC:
[NFC Helper Script](https://gist.github.com/lyellick-dev/68787d37b10bb7f17e11deddcfe9f974) 
Goals: Web NFC tag inventory app/Web NFC JS Library

So, thank you for the hard work! Not sure who else to send the good vibes to, but at least this can get to the community. 

-- Lincoln

Fortuntato, 2020-05-12:
I am completely new to this type of project environment and I was wondering if you could give me some guidance to see this kind of work with Web-NFC working. Is there anywhere I can have a play around to see your work? I am trying to find a demo where I can understand this technology more in depth

riju, 2020-05-12:
https://web.dev/nfc/  should be a good reference.

lyellick, 2020-05-12:
Yeah @riju posted the best starting point. 

My biggest hurdle at first was just getting the right browser and enabling the correct flags.

1. Download chrome dev
2. In mobile browser go to chrome://flags and enable Experimental Javascript and Experimental Web Platform Feature

Then after that i was having issues getting the permission prompt to appear, but I think this javascript class i wrote has things covered pretty well as a starting point: https://gist.github.com/lyellick/68787d37b10bb7f17e11deddcfe9f974

You can email at lincoln.yellick@gmail.com if you run into any issues.

lyellick, 2020-05-19:
@Fortuntato Just checking in to see if you were able to get started.

kenchris, 2020-05-20:
Thank you for the kind words @lyellick - closing for now!

closed 2020-05-20 by kenchris