w3c-cg/web-nfc #557: Shall we obtain and compare the ScanOptions?

DonnaWuDongxia, 2020-04-07:
In the spec:
https://w3c.github.io/web-nfc/#the-ndefreader-and-ndefwriter-objects

> Note that the internal slots of NDEFReader come from the options passed to NDEFReader.scan(). Therefore there is maximum one filter associated with any given NDEFReader object and successive invocations of NDEFReader.scan() with new options will replace existing filters.

```
dictionary NDEFScanOptions {
  USVString id;
  USVString recordType;
  USVString mediaType;
  AbortSignal? signal;
};
```
It aims to replace the filters for a reader. The question is what the expection if the |options| is identical or equals? Reject the successive _Scan_? And what if the signal is identical but other memebers not equals?

So the phase 

> with **new** options
is not eaily to identify.

Can we just define the action as replaceing the previous filter no matter the successive invocations of NDEFReader.scan() takes? The previous scan will onerror, or the signal passed to previous scan invocation will be aborted if there is. Just like SignalAbort on current scan + A new Reader.Scan().

zolkis, 2020-04-07:
The intent was the new filter (at scan invocation) replaces the internal slot entirely (no comparison) and when scan() needs it, will get that one from the internal slot.

DonnaWuDongxia, 2020-04-10:
Thanks for explanation @zolkis, and there still is a unclear point here for the "filter". Does the "filter" include |signal| field?
A CL https://chromium-review.googlesource.com/c/chromium/src/+/2145263 has been uploaded to support successive scan().
It makes the previous signal functional if no successive signals passed. Is this OK?
```
nfc_test(async (t, mockNFC) => {
  const reader = new NDEFReader();
  const controller = new AbortController();
  const scanOptions1 = {recordType: "url", signal: controller.signal };
  const scanOptions2 = {recordType: "mime"};

  await reader.scan(scanOptions1);
  const promise = reader.scan(scanOptions2);
  controller.abort();

  await promise_rejects_dom(t, 'AbortError', promise);
}, "Aborting on previous signal can stop current reading if no new signals \
passed to successive scan().");
```

zolkis, 2020-04-13:
For a given [NDEFReader](https://w3c.github.io/web-nfc/#dom-ndefreader), its [scan() algorithm](https://w3c.github.io/web-nfc/#the-scan-method) says:

[image]

In 4.1., if a new signal is specified, replace the old one.
If not specified, keep the old one.
In step 5, if the signal is aborted (even if the old one), reject the promise.
In step 6, if there is a signal, set up the signal abort steps.
So yes, according to these steps, if the new filter does not specify a signal, it inherits the old one.

So @DonnaWuDongxia you are correct, but @kenchris , is this really what was meant?

I find this inconsistent, as I would expect that if the new filter does not specify a signal, then there is no signal in the new scan, since the signal is part of [NDEFScanOptions](https://w3c.github.io/web-nfc/#dom-ndefscanoptions). (Unless there is some convention about signals that I miss here).

Also, note that in step 7.4, the reader is attempted to be added _again_ to the [activated reader list](https://w3c.github.io/web-nfc/#dfn-activatedreaderlist), which is a set, which does not allow duplicated and ignores adding if the item already exists in the set. So we only update the internal slots of the reader.

[image]

kenchris, 2020-04-14:
I believe we agreed just before the origin trial that we wanted to get rid of the filtering. It complicates a lot and its hard to explain to users how it actually works because it doesn't really filter.

@beaufortfrancois @zolkis

zolkis, 2020-04-16:
It would be nice to have some statistics from the origin trial how much filters have been used.
I can imagine little use, in which case we can deprecate them.

kenchris, 2020-04-22:
Filed https://github.com/w3c/web-nfc/issues/560

closed 2020-05-20 by kenchris