w3c-cg/web-nfc #558: Read only feature

jicihome, 2020-04-16:
Hi!
I was wondering how can we "lock" the NFC tag after writting or make it read only? I see that this some security specification, but didn't found a way to do it using Write or another method.

Thanks!

pechisworks, 2020-05-09:
This would also be useful for my use cases.

nico-martin, 2021-08-22:
I'm currently playing around with this and I think this is a crucial feature. Let's imagine I have NFC Tags in a Museum and anyone would be able to overwrite the content. That would be quite bad.

But looking through the Android specification I've seen NFC as a standard already has such a feature. So if i write an NFC Tag with readOnly from an Android app, I guess WebNFC could not just overwrite that content, right? If so, this would be a workaround for now..

beaufortfrancois, 2021-08-30:
@kenchris What are your thoughts?

Thanks for the feedback @nico-martin !
I'd love to see more signals from developers requesting this feature before adding it to Web NFC.

```json5
[SecureContext, Exposed=Window]
interface NDEFReader : EventTarget {
  ...
  Promise<undefined> makeReadOnly()
};
```

Note that this addition would be pretty simple indeed on [Android]( https://developer.android.com/reference/android/nfc/tech/Ndef#makeReadOnly()), but not necessarily on [iOS currently](https://github.com/chariotsolutions/phonegap-nfc/issues/409).

kenchris, 2021-08-30:
I am totally fine with this! It can return a promise and just reject if not supported

nico-martin, 2021-08-30:
Maybe a general NFC-question: How can I revoke the readOnly flag? Is this possible with NFC Tags?

beaufortfrancois, 2021-08-30:
This NFC operation is a one-way process and cannot be reverted for NFC tags.

nico-martin, 2021-08-30:
Ok. Maybe this could lead to some confusion for people that are not familiar with this.
Would it make sense to explicitly require a flag from the user that indicates this one-way usage?
This would be similar to the Web Blutooth API, where `requestDevice` explicitly requires an `acceptAllDevices: true` if no filters are set.

```javascript
const ndef = new NDEFReader();
try {
  await ndef.write({
    records: [{ recordType: "url", data: "https://w3c.github.io/web-nfc/" }]
  });
  await ndef.makeReadOnly({
    allowIrreversibleOperation: true
  });
} catch (e) {
  console.log("Something went wrong.", e);
};
```

kenchris, 2021-08-30:
The Chrome UI could show trusted UI explaining the risk and allowing the user to cancel or proceed

nico-martin, 2021-08-30:
Sure. That makes sense.

beaufortfrancois, 2021-11-29:
I've started a spec PR to address web developers need to make NFC tags read only at https://github.com/w3c/web-nfc/pull/632. Thoughts are welcome!

beaufortfrancois, 2021-12-08:
And I've just sent the "intent to prototype" in Chromium: https://groups.google.com/a/chromium.org/g/blink-dev/c/iyljeGnIct8

beaufortfrancois, 2021-12-15:
FYI Making an NFC tag read-only is now possible in latest Chrome Canary for Android:

1. Enable "experimental web platform features" in chrome://flags
2. Go to https://make-read-only.glitch.me and enjoy ;)

Feedback is always welcome!

See https://twitter.com/quicksave2k/status/1471060640131436546

beaufortfrancois, 2022-02-01:
This feature is shipping in Chrome 100. 
See https://twitter.com/quicksave2k/status/1488430972748111873

closed 2022-02-01 by beaufortfrancois