w3c-cg/web-nfc #559: Support TNEP

markus-becker-tridonic-com, 2020-04-22:
From https://nfc-forum.org/product/nfc-forum-tag-ndef-exchange-protocol-tnep-candidate-specification-1-0/: 

> The TNEP 1.0 Technical Specification supports the bi-directional exchange of NDEF messages based on the communication protocol used by the NFC Forum Tag devices of Type 2, 3, 4 and 5. The new TNEP protocol offers a simple protocol for NFC IoT devices to exchange data between an NFC enabled phone and the IoT Device. For example, this protocol can be used to configure and read smart meter devices, to control the thermostatic radiator valve or to configure the lightning device in your smart home.

Would be nice to support this with web-nfc and/or a library on top of web-nfc.

Devices based on Nordic chips implement TNEP tag-wise:
https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/include/nfc/tnep/tnep.html

kenchris, 2020-04-22:
Is this a replacement for SNEP which is being deprecated in Android?

NFC Simple Exchange Protocol
===
The SNEP is a simple protocol for doing peer-to-peer (P2P) NDEF transfers. It is not supported by CoreNFC (iOS) and it is being deprecated on Android (where is branded Android Beam) due to security issues).

markus-becker-tridonic-com, 2020-04-22:
I do not know much more that what is written in https://futureiot.tech/nfc-forum-releases-new-specifications-to-improve-connectivity-of-iot-devices/.

zolkis, 2020-04-22:
Stack explained here:
https://nfc-forum.org/our-work/specification-releases/specifications/#specs

kenchris, 2020-04-22:
https://www.businesswire.com/news/home/20190612005060/en/NFC-Forum-Enhances-Connectivity-IoT-Devices-New

I guess we need some native (Android etc) support before we can consider supporting this?

zolkis, 2020-04-22:
We have not exposed SNEP directly, so I wonder do we have to expose TNEP directly, or can it be encapsulated? What TNEP specific interactions or options need to be exposed to web pages?

And yes, a prerequisite is Android support for this.

markus-becker-tridonic-com, 2020-04-22:
> https://www.businesswire.com/news/home/20190612005060/en/NFC-Forum-Enhances-Connectivity-IoT-Devices-New
> 
> I guess we need some native (Android etc) support before we can consider supporting this?

Isn't it layered on top of NDEF without the need of P2P?

kenchris, 2020-04-22:
From my quick look at this, this looks like an official HCE (host card emulation) API, so it will probably need something like that. I think it is great with an official HCE API but it requires Android moving to this new API or us to emulate it ontop of what is already there.

This probably requires a lot of research

kenchris, 2020-04-22:
New records to support:

[image]

beaufortfrancois, 2020-04-22:
What is the platform support of those records?

kenchris, 2020-04-22:
I don't think you need any platform support for those as long as you can write well-known records, like smart poster without native support - can we do that?

kenchris, 2020-04-22:
As far as I understand - only Tag Type 2, 3, 4, 5 have been updated to support TNEP:

[image]

I assume this means that any TNEP Polling Device can only read these tag types (not non-standard like Mifare, or NFC Forum Type 1) and Android HCE (which emulates Type 4)

kenchris, 2020-04-22:
From reading source code: 

```
#define NFC_TNEP_VERSION 0x10
enum nfc_tnep_comm_mode {
	/** Single response communication mode */
	NFC_TNEP_COMM_MODE_SINGLE_RESPONSE,
	/** Service specific communication mode */
	NFC_TNEP_COMM_MODE_SERVICE_SPECYFIC = 0xFE
};
/** Maximum Service Waiting Time. */
#define NFC_TNEP_TAG_MAX_WAIT_TIME 63
/** Maximum Waiting Time extension. */
#define NFC_TNEP_TAG_MAX_N_WAIT_TIME 15
```

Service Parameters Record "Tp" payload:

byte: version (0x10)
byte: uri length in bytes
uri length bytes: uri
byte: communication_mode (0x01 single response, 0xFE service specific) 
byte: min_time (0...63 - converted to time units using protocol specified formula)
byte: max_time_ext (0...15 repetitions)
2 bytes (Big Endian): max_size in bytes

--
```
enum nfc_tnep_status_value {
	/** Success */
	NFC_TNEP_STATUS_SUCCESS,

	/** TNEP protocol error */
	NFC_TNEP_STATUS_PROTOCOL_ERROR,

	/** First service error code. */
	NFC_TNEP_STATUS_SERVICE_ERROR_BEGIN = 0x80,

	/** Last service error code. */
	NFC_TNEP_STATUS_SERVICE_ERROR_END = 0xFE,
};
```

TNEP status "Te" payload:

byte: status (0x01 success, 0x02 protocol error, 0x80...0xFE service specific errors)

--

Service Select "Ts" payload:

byte: uri length in bytes
uri length bytes: uri

kenchris, 2020-04-22:
It also seems like we could implement the TNEP Tag Device API on top of Android HCE (Host Card Emulation, Type 4 Tag emulation + ISO-DEP / APDU). We considered this already as a replacement for SNEP (NDEF push).

It is exactly how it is accomplished on the Nordic devices, if you check the Tag Device demo:

https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v12.2.0%2Fgroup__nfc__t4t__lib.html

kenchris, 2020-04-22:
Split the Tag Device part out into https://github.com/w3c/web-nfc/issues/563

markus-becker-tridonic-com, 2020-04-27:
Would it be possible to test TNEP with WebNFC right now using external type records or would one have to wait until Tp, Te and Ts have been added to the specification and implementation?

kenchris, 2020-04-28:
@markus-becker-tridonic-com you would have to wait until that is added yes.

kenchris, 2020-04-28:
Did some exploration here: https://github.com/w3c/web-nfc/issues/567