w3c-cg/web-nfc #560: Get rid of filtering + AbortController

kenchris, 2020-04-22:
It turns out that filtering was not the greatest idea

* it is a bit confusing as it is not filtering the records (not sure that would actually be that useful without complex rules) and it just filters whether such a record exist 
* Newer standards like TNEP have other ways of selecting services as a way of filtering/communicating with IoT devices.

So let's get rid of it and simplify spec and implementation. @zolkis 

Also, lots of thought around AbortController with TAG members.

`scan` is a continuously running operation until cancelled, (or loss of focus etc) and not just a long running task, so me and others have come to the conclusion that AbortController is not the proper way to cancel our scans. I suggest that we follow Generic Sensors instead and introduce a `start()` and `stop()` method instead.

beaufortfrancois, 2020-04-22:
I would split this issue in two: "Get rid of filtering" and "Get rid of AbortController".
Regarding the AbortController, can you share the TAG members notes?

zolkis, 2020-04-22:
I agree with all the above.
@leonhsl are you fine with this?

kenchris, 2020-04-22:
> I would split this issue in two: "Get rid of filtering" and "Get rid of AbortController".
> Regarding the AbortController, can you share the TAG members notes?

https://github.com/w3c/web-nfc/issues/562

We didn't minute that, but I am assigned to add that policy to our design policy document but haven't gotten around to it yet

kenchris, 2020-05-20:
Leave open until merged to spec

kenchris, 2020-05-25:
Done on branch: https://github.com/w3c/web-nfc/commit/c6eaae294626ab2f59276d54367b83094b57a6ff

kenchris, 2020-05-25:
Looking through the code, I don't see any where the filtering is actually done, so it might be as easy to remove the 3 properties in the dictionary in the files below:

third_party/blink/web_tests/external/wpt/interfaces/web-nfc.idl
third_party/blink/renderer/modules/nfc/ndef_scan_options.idl

@zolkis is this handled in the spec?

[image]

zolkis, 2020-05-25:
> @zolkis is this handled in the spec? 

In the draft branch, yes, by #565

beaufortfrancois, 2020-05-26:
Filtering is done on the Java side:
https://source.chromium.org/chromium/chromium/src/+/master:services/device/nfc/android/java/src/org/chromium/device/nfc/NfcImpl.java;l=615;drc=cc291e324ffba9c0282d62f00c2005637d5fb37a?originalUrl=https:%2F%2Fcs.chromium.org%2F
https://source.chromium.org/chromium/chromium/src/+/master:services/device/nfc/android/java/src/org/chromium/device/nfc/NfcImpl.java;l=597;drc=7f2a66319b71e1cd3be556aeaa50cd18d1be70c1?originalUrl=https:%2F%2Fcs.chromium.org%2F

kenchris, 2020-05-26:
I see, seems easy to remove though, so that is good!

kenchris, 2020-12-08:
This has been integrated into the spec

commit 579613b9 (2020-04-29): Fix #560: remove filtering
commit c6eaae29 (2020-04-29): Merge pull request #565 from zolkis/remove-filtering

Fix #560: remove filtering
closed 2020-12-08 by kenchris