w3c-cg/web-nfc #561: Join Reader and Writer objects

kenchris, 2020-04-22:
I take the blame for this. It originally seemed like a good idea with separate objects, but I am getting more and more convinced that it was a bad idea.

* If we enhance to be able to select tag in the case of multiple tags read, this is required
* TNEP also requires this
* We can get rid of ignoreRead
* User is more in control of what happens.

Suggested name `NDEFPoller` or `NDEFPollerDevice`

OlivierGrenoble, 2020-04-23:
> If we enhance to be able to select tag in the case of multiple tags read, this is required

On Android, it is not possible to select one tag among several tags. There should be only one in the RF field.

reillyeon, 2020-04-23:
I vote for calling the combined object `NDEFReader`. The physical hardware components for NFC seem to usually be called a "reader" even though they can also write. It doesn't seem necessary to find a new (potentially awkward) word to describe it.

kenchris, 2020-04-23:
I need to get hand on the new NFC TNEP spec, but it looks like they used poller device (vs tag device) but I need to confirm

OlivierGrenoble, 2020-04-24:
> I vote for calling the combined object `NDEFReader`. The physical hardware components for NFC seem to usually be called a "reader" even though they can also write. It doesn't seem necessary to find a new (potentially awkward) word to describe it.

I agree

kenchris, 2020-04-24:
I checked the new spec and it seems that the Poller thing was just what Nordic named it in their API.

> Currently the following NFC Forum Devices are defined:
>  * NFC Universal Device
>  * NFC Tag Device
>  * NFC Reader Device.

zolkis, 2020-04-26:
> I vote for calling the combined object NDEFReader

> Currently the following NFC Forum Devices are defined:
NFC Universal Device
NFC Tag Device
NFC Reader Device

Ergo what about `NFCReader`, bearing in mind possible future support for NDEF+other tech for a given device?

reillyeon, 2020-04-27:
I'd be cautious about assuming that the existing API can be extended to support future technologies. Keeping the API focused on NDEF allows us to only consider a subset of all possible NFC interactions and built the best API for that.

zolkis, 2020-04-27:
Discussed with Kenneth that we'd likely stick with `NDEFReader` that will have a `write()` method :).

kenchris, 2020-05-25:
Fixed on branch: https://github.com/w3c/web-nfc/pull/568

kenchris, 2020-12-08:
Merged upstream

commit e5ecb1bb (2020-04-30): Fix #561: join NDEFReader and NDEFWriter (#568)
closed 2020-12-08 by kenchris