w3c-cg/web-nfc #562: Get rid of AbortController

kenchris, 2020-04-22:
Also, lots of thought around AbortController with TAG members.

`scan` is a continuously running operation until cancelled, (or loss of focus etc) and not just a long running task, so me and others have come to the conclusion that AbortController is not the proper way to cancel our scans. I suggest that we follow Generic Sensors instead and introduce a `start()` and `stop()` method instead.

kenchris, 2020-04-22:
> Regarding the AbortController, can you share the TAG members notes?

We didn't minute that, but I am assigned to add that policy to our design policy document but haven't gotten around to it yet.

As an alternative we could (and probably should as spec and impl is there) still allow AbortController as an alternative in the `start()` method. I am reading that they are adding that alternative to the *Stream constructor

lyellick-dev, 2020-04-22:
I like the idea of keeping the abort controller as an option since I am given more freedom to hook more events within the abort controller signal. Now saying that I remember when I initially started working with this I was supprised there wasn't just a "stop" method, but after understanding the abort controller and signal I appreciated it. 

Example of how I am using it [here](https://gist.github.com/lyellick-dev/68787d37b10bb7f17e11deddcfe9f974).

_Note: Please let me know if this the wrong place to put my input._

kenchris, 2020-05-04:
Even Web Bluetooth Scanning doesn't use AbortController and also has a stop method. 

https://webbluetoothcg.github.io/web-bluetooth/scanning.html#bluetoothlescan

It is a bit different than just start/stop, as the main web bluetooth objects does a lot more than scanning. Here you receive an object back that you can call `stop` on.

beaufortfrancois, 2020-05-11:
For info, https://github.com/WICG/idle-detection/issues/19

kenchris, 2020-05-11:
Thanks Francois. I guess we should just rename to start() then and we will be consistent with IdleDetector and future Generic Sensors

zolkis, 2020-05-11:
Works for me.

beaufortfrancois, 2020-05-11:
`start()` and `write()` look odd to me under the same NDEFReader interface.

zolkis, 2020-05-11:
True, a `start()` would have made sense in a separate `NDEFReader` interface, but with the merged objects it doesn't. `scan()` works for me, too.

kenchris, 2020-05-20:
Let's close this then

closed 2020-05-20 by kenchris