w3c-cg/web-nfc #563: Support TNEP: Tag Device (an alternative to deprecated push/SNEP)

kenchris, 2020-04-22:
When SNEP (simple NDEF Exchange Protocol) was deprecated we researched how to support sending NDEF to other active devices.

The conclusion was that we could implement it using Android HCE (Host Card Emulation) a non-standard for emulating Type 4 Tags with ISO-DEP/APDU. On top of that you could implement NDEF and basically emulate a Type 4 tag with NDEF. Pretty smart, but also a bit non standard.

Now with TNEP this seems to have become a more official way. TNEP allows to create a TNEP (Tag NDEF Exchange Protocol) Tag Device, which is like an emulated tag - thus basically HCE.

It is exactly how it is accomplished on the Nordic devices, if you check the Tag Device demo, which emulates a Type 4 Tag and uses the TNEP records on top.

https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v12.2.0%2Fgroup__nfc__t4t__lib.html

eklem, 2022-06-29:
Would this make phone to phone possible?

getify, 2022-10-16:
Would *very much* like this, [for use cases as explained here](https://github.com/WebBluetoothCG/web-bluetooth/issues/231#issuecomment-1279991680).

akbertram, 2023-06-20:
Came here looking for the peer-to-peer use case. Very interested in this for our PWA case management and survey app, which works well offline, but requires an internet connection to sync with a central server.

One use case that comes up periodically is teams working in remote areas with no network connection for extended period of time, particularly in humanitarian relief contexts. Typically one team member might register the househould, but another would conduct an assessment. With peer to peer NFC the caseworkers could exchange unsychronized drafts to build on each others drafts.

anssiko, 2023-06-21:
@getify @akbertram you may be interested in the new Local Peer-to-Peer API proposal we just submitted for consideration: https://github.com/WICG/proposals/issues/103

At this stage we’re looking for support signals and use cases to prioritize this exploration appropriately to ensure it meets the needs of web developers. Your use cases discussed in this issue would be much appreciated contributions to help that proposal move forward.

@getify where can we track your latest for the zero-server sharing exploration?

cc @ibelem

eklem, 2023-06-21:
I got a couple of user stories. The small one that will be part of other user stories is to transfer JSON from one phone to another, not going through online services. The bigger and more specific one is to transfer one time pads for encryption/decryption of messages. It can solve one of the big problems with one time pad encryption/decryption.

anssiko, 2023-06-21:
@eklem, those are great user stories! Please provide your support and user story input in https://github.com/WICG/proposals/issues/103 directly so the rest of the web community interested in new incubations can chime in. Thanks for your contributions! We'll then integrate your input to the explainer document in its next revision.

maceip, 2024-11-18:
4 years!

**what web developers and web users want**: mobile web nfc (+wifi +ble) that works across android / ios for the obvious use cases: p2p tap payment, data transfer, etc

what we get: bullshit from apple and tarpit w3c :(