w3c-cg/web-nfc #566: Parse NDEF T records: data set to full PAYLOAD?

zolkis, 2020-04-27:
In [Parsing T records](https://w3c.github.io/web-nfc/#parsing-ndef-well-known-t-records),

[image]

Step 10, we set `data` to the full PAYLOAD.

kenchris, 2020-12-08:
Has this been fixed?

beaufortfrancois, 2020-12-08:
Spec has **not** been fixed indeed. @zolkis Can you have a look?

zolkis, 2020-12-08:
Need to check the implementation as well.

beaufortfrancois, 2020-12-08:
Implementation is good. See https://source.chromium.org/chromium/chromium/src/+/master:services/device/nfc/android/java/src/org/chromium/device/nfc/NdefMessageUtils.java;drc=bf799475f9ff40b7e1e2be2fd3a68911c4f047ee;bpv=1;bpt=1;l=271

```java
    private static NdefRecord createTextRecord(byte[] text) throws UnsupportedEncodingException {
        // Check that text byte array is not empty.
        if (text.length == 0) {
            return null;
        }

        NdefRecord nfcRecord = new NdefRecord();
        nfcRecord.category = NdefRecordTypeCategory.STANDARDIZED;
        nfcRecord.recordType = RECORD_TYPE_TEXT;
        // According to NFCForum-TS-RTD_Text_1.0 specification, section 3.2.1 Syntax.
        // First byte of the payload is status byte, defined in Table 3: Status Byte Encodings.
        // 0-5: lang code length
        // 6  : must be zero
        // 7  : 0 - text is in UTF-8 encoding, 1 - text is in UTF-16 encoding.
        nfcRecord.encoding = (text[0] & (1 << 7)) == 0 ? ENCODING_UTF8 : ENCODING_UTF16;
        int langCodeLength = (text[0] & (byte) 0x3F);
        nfcRecord.lang = new String(text, 1, langCodeLength, "US-ASCII");
        int textBodyStartPos = langCodeLength + 1;
        if (textBodyStartPos > text.length) {
            return null;
        }
        nfcRecord.data = Arrays.copyOfRange(text, textBodyStartPos, text.length);
        return nfcRecord;
    }

```

commit dfb84005 (2020-12-08): Merge pull request #612 from zolkis/fix-text-records

Fix #566: parse text records
commit e9bcb3d3 (2020-12-08): Fix #566: parse text records
closed 2020-12-08 by zolkis