w3c-cg/web-nfc #569: Simplify scan()

zolkis, 2020-04-30:
After filtering is gone and we have separate readers each with internal slots, IMHO we could simplify scans:
- resolve [scan()](https://w3c.github.io/web-nfc/#the-scan-method)'s promise with a <[serialNumber](https://w3c.github.io/web-nfc/#dom-ndefreadingevent-serialnumber), [NDEFMessage](https://w3c.github.io/web-nfc/#the-ndefmessage-interface)> tuple
- remove the [`onerror`](https://w3c.github.io/web-nfc/#dom-ndefreader-onerror) and [`onreading`](https://w3c.github.io/web-nfc/#dom-ndefreader-onreading) events
- join the [scan](https://w3c.github.io/web-nfc/#the-scan-method), the [NFC reading algorithm](https://w3c.github.io/web-nfc/#the-nfc-reading-algorithm) and the [dispatching NFC content](https://w3c.github.io/web-nfc/#dispatching-nfc-content) algorithms.

kenchris, 2020-04-30:
No, let's not do that please - multiple reasons :-)

1. `scan` initiates polling and that is also needed for writing, so this would screw up letting people in control of what is happening and easily implement timeouts, ignore reads while writing etc.

It is also needed for TNEP communication, which is not a just-read-one-value and your are done mode.

2. That is needed as we are not doing 1. 

Not sure about 3. with the above answers.

With all of that said, today `write()` has a bit of magic which might be fine for the simple case, ie. it basically does what you suggest `scan()` would do above.

Another option would be doing similar (and similar to what we wanted to do for GeolocationSensor which got abandoned for now) - ie adding the convenience of a `read()` method that does a single read. Then in return, rename `scan()` to `start()` and add a `stop()`

zolkis, 2020-05-01:
> adding the convenience of a read() method that does a single read.  

This crossed my mind, too. 
We could leave scan() as it is with AbortController (for which we received dev feedback to keep).

Anyway it's a convenience and someone can implement it with scan(), so I am not sure it's worth doing. Maybe we could include an example on how to implement a simple read() that resolves with the data.adding the convenience of a read() method that does a single read?

kenchris, 2020-05-01:
That example exists in my latest PR

closed 2020-05-04 by zolkis