w3c-cg/web-nfc #573: Encoding Standard stream dependency needs work

annevk, 2020-05-06:
https://github.com/whatwg/encoding/issues/180 was never resolved, but a PR still landed here...

zolkis, 2020-05-07:
It became irrelevant since #265, as the [push](https://encoding.spec.whatwg.org/#concept-stream-push) concept is not used any more from the Encoding Standard.

kenchris, 2020-05-07:
@zolkis there is still wording around push in the spec. Would make sense to look over all of that and update it. Make it clear that this is not being pursued and why

zolkis, 2020-05-08:
I checked before commenting and ATM there is no more reference to encoding/push.
The word "push" figures once in a different context.

zolkis, 2020-05-08:
But there are other stream dependencies, the concept, read, end of stream, but they seem to be okay.

annevk, 2020-05-08:
No that's not okay, see https://github.com/whatwg/encoding/issues/180#issuecomment-518155671 onward.

zolkis, 2020-05-08:
Ah, I see. We'll keep this open. Thanks Anne.