w3c-cg/web-nfc #578: APDU transmit

OR13, 2020-05-23:
Hello, I'm testing with origin trial, and trying to reproduce this working node.js sample:

https://github.com/OR13/nfc.did.ai/blob/master/packages/tangem-sdk-node/src/APDU/readCard.ts

Note the `reader.transmit`... https://github.com/pokusew/nfc-pcsc#reading-and-writing-data

This is needed for raw APDU communication: https://en.wikipedia.org/wiki/Smart_card_application_protocol_data_unit

Is there support for non-NDEF transmission?

kenchris, 2020-05-25:
There is not at the moment no, and not for at least the initial release.

Fortuntato, 2020-06-21:
Hi there,
I think it would be an useful enhancement since I was thinking to do something once the web app understand the smart card origin. For example a simple use case would be:

An online shop has student discounts and needs a valid student verification.

- The Web App can prompt a page that ask to tap with the student ID on the phone
- Reading the smart card it will find out the University name and check if that is a valid student

Of course it will need to do many third party checks but it could be an useful feature as form of validation.

I guess it could be done on a mobile application too but I think it would be less used.

OR13, 2020-06-22:
Yes, especially useful for smart cards that can produce digital signatures or encryption / decryption using hardware isolated keys. 

Web AuthN Lists NFC as a transport:

https://www.w3.org/TR/webauthn/#transport

This was my original reason for investigating NFC.

I was able to work around the limitation by creating a "Kiosk Setup", where an NFC Reader is connected to a web server and the web server is connected to a website (with or without CORS).

This also lets you use NFC from a desktop browser, which is very convenient for development.

OR13, 2020-11-13:
@kenchris @zolkis @beaufortfrancois 

I know this has already been asked and answered, but this feature would be complete game changer.

There are a lot of NFC products out there that don't speak NDEF.

If there is anything I, as a developer / W3C member can do to help get a lower level interface in please let me know.

hmiguel, 2021-03-02:
> @kenchris @zolkis @beaufortfrancois
>
> I know this has already been asked and answered, but this feature would be complete game changer.
>
> There are a lot of NFC products out there that don't speak [...]

Agree, NDEF is in fact pretty useless these days.

zolkis, 2021-03-02:
Yes, we did think about extending, and there are `brainstorming` labeled discussions in the repo on the possible ways forward.
However, let's pull NDEF first.
Then, from this point forward, both the API spec and Chrome on Android implementation need more contributors. 
If your companies are not W3C members, there is a possibility to become Invited Experts. @anssiko ?

anssiko, 2021-03-02:
Given this work happens in the [Web NFC Community Group](https://www.w3.org/community/web-nfc/), getting permission to contribute materially (proposing concrete spec changes, submitting PRs etc.) to the Web NFC API specification is even easier - just join the Community Group which is _open and free to attend for all_.

@OR13 and @hmiguel, and others interested in helping move this or other features forward, please see the above link and click join.

Gimly-Caspar, 2021-04-28:
This will be highly valuable indeed. We are also looking to implement ECC capable smartcards in web applications.

emirhanai, 2023-01-13:
#APDU is very need to projects. @Gimly-Caspar @hmiguel @zolkis @anssiko @kenchris @OR13

anssiko, 2023-01-13:
Thank you for your continued interest and support. We welcome further use cases in this issue on how this feature would be used by real-world web applications. Use cases can be contributed by anyone, even without joining the group.

All feedback will be considered in planning future work for this API. Generally, use cases with a clear end user benefit will help raise feature priority. That guideline applies for this API as well.

mitar, 2023-04-05:
In Slovenia, we have now new [biometric identity cards](https://www.gov.si/en/news/2023-03-08-slovenian-biometric-identity-card-wins-prestigious-international-award/) which support signing and authentication through NFC and APDU. It would be great if web apps could use them to authenticate and sign documents without having to have additional software installed. (Currently one has to install a mobile app which then more or less acts just as a NFC reader proxy app.)

OR13, 2023-04-05:
Doesn't mDoc use NFC? or does that only work on Apple devices?

ManuelB, 2023-04-29:
Hi, I would need this to "talk" to a german health card and for example read a medication plan.