w3c-cg/web-nfc #589: Web NFC available on Chrome 84?

Fortuntato, 2020-07-05:
Hi all, as we are approaching the date when Chrome v.84 will be released I was wondering if this feature is going to be released or not because, long story short, I have made my University dissertation about this feature and I have produced a Web App with this feature. 
Once I complete it I would be more than happy to share my research with you. But for the moment I really need to know what happens on the 14th July when the new version of Chrome is released because that is a crucial date for me if something changes.

riju, 2020-07-06:
@Fortuntato : As you might be aware, we did run Origin Trials and we got a few feedback from developers. We are addressing those them right now.

Fortuntato, 2020-07-06:
@riju Thank you for the reply. So for the time being until the end of the month, the feature will be still available as part of the Experimental Web Features flag on Chrome?

beaufortfrancois, 2020-07-06:
Yes. The flag will still be available in Chrome.

anssiko, 2020-07-06:
@Fortuntato I'm excited to hear you chose Web NFC as your dissertation topic. Please contribute your findings to the group when you're ready to do so. Also, I encourage you to [join the group](https://www.w3.org/community/web-nfc/) to keep in touch even after you're done with your dissertation. Maybe you'll be using this technology in your future projects.

I'll close this issue now. Please chime in if you have any further questions.

closed 2020-07-06 by anssiko

Fortuntato, 2020-07-26:
Hi all! I have finalised my software project and dissertation/report about it. This is my project repository [https://github.com/Fortuntato/Contactless-Voucher](url).
A copy of the project report can be downloaded here: [Documentation/Final Report/Contactless Loyalty Project Report.pdf](url)