w3c-cg/web-nfc #592: Decide whether multiple scanners are allowed

kenchris, 2020-07-10:
Currently as spec'ed each reader instance will receive the same data.

I am not a bit fan of this. iOS allows the same, but there is an order to it, like the first (or maybe it was the last) added reader receives the data first and if it doesn't handle it, it will be given to the next reader. That makes more sense to be, but this is the web so we really don't want other libraries etc to do any reading, so I think it makes more sense for the call to `scan` to actually cancel existing scans.

@riju @beaufortfrancois

riju, 2020-07-10:
Context: Multiple scans in the world where there is no filtering.
We added support for multiple scan invocations on the same Reader in this [CL](https://chromium-review.googlesource.com/c/chromium/src/+/2145263)

`NDEFScanOptions/signal` is there to abort a particular scan. Calling `scan` to cancel other scans is a bit odd ?

beaufortfrancois, 2020-07-10:
I'm looking at other APIs and how they deal with multiple operations with an AbortSignal.

The Idle Detection API doesn't allow starting a second time when it's already started
```
Uncaught DOMException: Failed to execute 'start' on 'IdleDetector': Idle detector is already started.
```

Something similar happens for Web Bluetooth and `watchAdvertisements`
```
Failed to execute 'watchAdvertisements' on 'BluetoothDevice': Pending watch advertisements operation.
```

So we could simply reject scan promise when there's already an ongoing scan. WDYT?

```js
const reader = new NDEFReader();
await reader.scan();
reader.scan(); // <- rejects immediately with InvalidStateError

kenchris, 2020-07-10:
I would be OK with that

zolkis, 2020-07-14:
Or new scan (likely initiated by the user) canceling the existing ones.
Looks better than telling the user "please cancel your previous scan manually". 
No strong opinion, just consider the best flow.

taralx, 2020-07-21:
The only question I have is if this will be a case of "which of my many tabs is holding the NFC scan". Might be ok if the UA has a persistent indication that allows the user to jump to the page though.

beaufortfrancois, 2020-12-08:
@kenchris We've decided to simply reject `scan` promise with InvalidStateError when there's already an ongoing scan.
However spec was not updated despite implementation changes at https://chromium-review.googlesource.com/c/chromium/src/+/2297559 FYI @riju 
@zolkis Do you mind updating the spec to address this issue?

zolkis, 2020-12-08:
NP

beaufortfrancois, 2020-12-16:
@zolkis See https://github.com/w3c/web-nfc/issues/592#issuecomment-740542712. Chromium implementation does not support multiple scans on the **same** reader, but it's fine for multiple readers. Please update spec with a `has_pending_scan_request_` bit as in https://chromium-review.googlesource.com/c/chromium/src/+/2297559/5/third_party/blink/renderer/modules/nfc/ndef_reader.cc

zolkis, 2020-12-16:
@beaufortfrancois 

Does it mean we deliberately make scan behavior different from write behavior?
Writes cancel previous writes. 

It seems you proposing the opposite for reads, i.e. that new scans should not replace previous scans, but fail with the error?
We seem to have implementation for this, whether or not it is the correct, or desired behavior.
But this would prevent the issue of how to notify about stopping event delivery for ongoing scans (we don't have a notification implemented yet). Although we could use the `onreadingerror` event to notify about scans being aborted (for any reason).

In my view, both are doable, we need to select one with good reasoning and we need to explain in a note why. 
Please suggest a text for that.

About multiple reads, the API design with read events works with both single and multiple reads, regardless of what the implementation supports, which is a good thing.

commit 59624e0f (2020-12-28): Merge pull request #611 from zolkis/fix-multiple-scanners

Fix #592: allow multiple readers but no multiple scans on any reader
commit 6dc7be83 (2020-12-28): Fix #592: scan rejects prior scan
closed 2020-12-28 by zolkis