w3c-cg/web-nfc #594: NFC not working on new version chrome

kaissair, 2020-07-29:
Hi, since 3 months now I use NFC API by and for google chrome, and today it's not working.
it is very disturbing because I use this API  for my system and myu users are block
Do you have any informations about re-activation afi this api by google or how can I active it again

Many thank's for your response
best regards

riju, 2020-07-29:
We are implementing the Origin Trial feedback and should expect to complete by M87. In the meantime, please enable the `#experimental-web-platform-features` flag in `chrome://flags`

kaissair, 2020-07-29:
Hi

Many thank's it's working now.

You make à Great job

👍👏👏👏

LEMMOUCHI Kaïss Jamil
33, Avenue de Wagram - 75017 Paris
Mob : +33 6 46 86 45 26
           +33 6 20 27 90 28
Mail : Kaiss@airgos.aero

echavezvise, 2020-08-03:
Thank you very much @riju, it was very helpful, I hope the project continues and will soon be released.

In the company I want to use it, in 3 web applications, 2 to read the tags and register them.
Another to read cargo trucks, and write the material and specifications since where we work we are offline.

Regards.

kenchris, 2020-12-08:
Sounds like this is resolved, also not a spec issue

closed 2020-12-08 by kenchris