w3c-cg/web-nfc #595: Chrome, nfc-list and ACR122

glococo, 2020-08-11:
Hi everyone.
I want to work in desktop with the ACR122 nfc pad but It seems to not be detected by Chromium, Chrome (Beta/Dev) in Ubuntu.

My current setup seems to work from console:

```guillermo@lab:~$ nfc-list -v
nfc-list uses libnfc 1.7.1
NFC device: ACS / ACR122U PICC Interface opened
1 ISO14443A passive target(s) found:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  44  
* UID size: double
* bit frame anticollision supported
       UID (NFCID1): 04  98  f9  ba  53  65  80  
      SAK (SEL_RES): 00  
* Not compliant with ISO/IEC 14443-4
* Not compliant with ISO/IEC 18092

Fingerprinting based on MIFARE type Identification Procedure:
* MIFARE Ultralight
* MIFARE Ultralight C
Other possible matches based on ATQA & SAK values:

0 Felica (212 kbps) passive target(s) found.

0 Felica (424 kbps) passive target(s) found.

0 ISO14443B passive target(s) found.

0 ISO14443B' passive target(s) found.

0 ISO14443B-2 ST SRx passive target(s) found.

0 ISO14443B-2 ASK CTx passive target(s) found.

0 Jewel passive target(s) found.

guillermo@tfe152:~$ pcsc_scan 
Using reader plug'n play mechanism
Scanning present readers...
0: ACS ACR122U 00 00
 
Tue Aug 11 11:44:52 2020
 Reader 0: ACS ACR122U 00 00
  Event number: 5
  Card state: Card inserted, 
  ATR: 3B 00

ATR: 3B 00
+ TS = 3B --> Direct Convention
+ T0 = 00, Y(1): 0000, K: 0 (historical bytes)
 \   
guillermo@lab:~$ ```

In Chrome I get in web-nfc sample:
```
User clicked scan button
Argh! NotAllowedError: NFC permission request denied.
```

Any help is really appreciated.

cyberphone, 2020-08-12:
Web NFC is only supported on Android.  NFC is no longer a default component of a PC (according to Intel).  QR is nowadays the recommended solution for interacting with an external device and the desktop Web.

zolkis, 2020-08-12:
Indeed, the current implementation of Web NFC is for Chrome on Android.

There has been some work done to use NFC middleware on Linux to implement Web NFC (e.g. using [neard](https://git.kernel.org/pub/scm/network/nfc/neard.git/)) but in the end we have focused on doing the first implementation on top of the more mature Android NFC API.

That being said, the doors are open for other Web NFC implementations. It would be really nice if someone would pick up that work (_neard_ maintenance and eventually Web NFC implementation on top of it).

kenchris, 2020-12-08:
Closing as this is not spec related

closed 2020-12-08 by kenchris