w3c-cg/web-nfc #596: Web NFC in iOS Chrome

vinisha246, 2020-08-18:
Hi, 
I have tried Web NFC in Android Chrome and it is working fine.
Does it work in iphone too? I have tried setting up, but **couldn't find "Experimental Web Platform features"** to enable.

anssiko, 2020-08-18:
See https://caniuse.com/#feat=webnfc for the current browser support.

There's a WebKit bug https://bugs.webkit.org/show_bug.cgi?id=163196 that has been open for a while.

See also https://web.dev/how-to-file-a-good-bug/ for general guidance on how to tell browser vendors about issues you discover.

closed 2020-08-18 by anssiko

manuelfink, 2020-11-23:
As I guess more are going to search for this. 

W3 Web-nfc standard likely is the wrong place to request this, thus I forwarded the question to Google Chrome CRBUG Tracker https://bugs.chromium.org/p/chromium/issues/detail?id=520391#c220

This is a duplicate of https://github.com/w3c/web-nfc/issues/230.