w3c-cg/web-nfc #602: Attach to navigator

zolkis, 2020-10-13:
This is to record the arguments in a discussion about merging NDEF reader and writer objects and removing filters, coming from Origin Trials feedback.
That resulted in using only one NDEF reader and one writer promise being part of NFC state (per settings object).
It means we don't need multiple NDEFReader objects, since only the one owning the last called scan() will be active at a time, the previous ones need rejected (with AbortError).
That in turn popped the question whether should we attach to the Navigator object, as Web Bluetooth, Web USB etc are doing.
Our explainer listed arguments for using NDEFReader() constructors, since it allows implementing the same API in Node.js, and it is aligned with newer APIs such as Generic Sensor.

On the other hand, Web NFC is closer to Web Bluetooth and Web USB than to Generic Sensors, therefore the Editors decided to attach to the Navigator object and use `navigator.nfc.ndef` of type NDEFReader (that would handle scans and writes).

In the future, when other than NDEF tech might be supported, they could be hosted under `navigator.nfc` as well.

zolkis, 2020-10-13:
Next, the question is how to attach to the `navigator`.

1. use `navigator.nfc.ndef` to represent an NDEFReader, which exposes `scan()`, `write()`, `onreading`, `onreadingerror`.
`onreading` will contain an `NDEFMessage`.

2. `navigator.nfc` to contain abortable `scan()` (which initiates NFC polling) and `ontagfound` event (basically the same what `onreading` is used for now) which presents an NDEFReader `ndef` that exposes async `read()`, `write()` [and eventually `format()` which is currently included in `write()`]. Later it could contain a control object for other NFC tech (as well as NDEFReader). Also, other events could be added later.

In both cases, `navigator.nfc` could contain  a status (`active` or `suspended` flag), and an `onvisibilitychanged` event. This is to reflect the TPAC 2020 discussion with WICG/Web Bluetooth people.

I'd vote for option 2 for a more correct and more future-proof API which is also better aligned with Andoid NFC, even if it's a bigger change than doing option 1. The current API conflates polling (scanning) and read functionality, which might be confusing to people coming from native NFC APIs.

zolkis, 2020-10-14:
Let's have some JS examples for option 2 above. [Example 4]() has to use `ignoreRead`, with option 2 `ignoreRead` is not needed.

```javascript
navigator.nfc.scan();  // start NFC polling
navigator.nfc.ontagfound = event => {
  if (event.ndef) {
    event.ndef.write(data).then({
      console.log("Tag written");
    });
    ndef.read().then( message => {
    });
  }
}
```

kenchris, 2022-06-29:
Closing as this isn't  going to change now we have shipped