w3c-cg/web-nfc #613: Every scan results in empty records

mrquincle, 2020-12-16:
Hi, I have tried half a dozen different NFC tags and cards. 

```
User clicked scan button
> Scan started
> Serial Number: 04:4a:2a:fa:5e:67:80
> Records: (1)
NDEFRecord {recordType: "empty", mediaType: null, id: null, encoding: null, lang: null, …}
data: null
encoding: null
id: null
lang: null
mediaType: null
recordType: "empty"
__proto__: NDEFRecord 
```
They all result in the `empty` recordType. If I just use an Android tag reader there's quite a lot of information in the NDEF records, so this seems incorrect. 

Tried on (of course with experimental web platform features enabled):
+ Chrome: 87.0.4280.66
+ Chrome: 87.0.4280.101
+ Chrome Canary: 89.0.4357.0 

I've run this on a OnePlus and a Lenovo phone.

Just using the sample code on https://googlechrome.github.io/samples/web-nfc/ (just added printing the records as well, to the console).

I don't think this has to do with the hardware (due to the fact that a utility like "NFC Tools" can easily read the memory. If there has been a regression, would be nice to know!

beaufortfrancois, 2020-12-16:
This seems like a browser issue, not related to spec directly. @mrquincle Can you file bug at https://bugs.chromium.org/p/chromium/issues/entry?components=Blink%3ENFC and share it with me so that we can diagnose the issue?

mrquincle, 2020-12-16:
You will always see, I got triggered by the commit where a lot of use cases that are not supported by Web RTC are removed or described as such. 

Hence, I found another NFC device and tried it as well. Now I get a proper `url` recordType, sweet! I'll close this. Sorry for the noise!

closed 2020-12-16 by mrquincle

beaufortfrancois, 2020-12-16:
Glad to know it's working for you ;)