w3c-cg/web-nfc #616: Web NFC Documentation Request: This site can’t ask for your permission Close any bubbles or overlays from other apps. Then try again.

lyellick, 2021-04-09:
This is not a bug in Web NFC, but a documentation gap. 

I could not enable a NFC permission to my site while having a "bubble" overlaid on my android device.  It took me a bit to figure out that the small YouTube preview on the bottom right of the screen was the issue. When closing the preview or "bubble" I was able to allow the NFC permission for the site without the notice there was an overlay present. I would recommend that in the documentation there was an addition to troubleshooting with a header as "This site can’t ask for your permission Close any bubbles or overlays from other apps. Then try again." and the explanation of closing any overlays such as YouTube or other apps that allow an overlay preview. The description of the error was not self explanatory to me so I believe others searching for this would appreciate the SEO of "Web NFC This site can’t ask for your permission Close any bubbles or overlays from other apps. Then try again." within the documentation of Web NFC. 

**Device**: Android Pixel 3 Version 11 Build RQ2A.210305.006, Chrome 89.0.4389.105

**Steps to show error:**
- Insure there is an app overlay such has YouTube showing when in Chrome.
- In Chrome navigate to: https://nfc.kerberus.io/
- Click "NFC Switch"
- Dialog: Enable NFC on site.
- Dialog: Allow
- Dialog: “This site can’t ask for your permission Close any bubbles or overlays from other apps. Then try again.”

beaufortfrancois, 2021-04-12:
Thank you @lyellick!

For reference, here's a screenshot of the issue.
[image]

I've filed https://bugs.chromium.org/p/chromium/issues/detail?id=1198082 to track this issue.
Feel free to comment over there and star it to be updated.

closed 2021-04-12 by beaufortfrancois

lyellick, 2021-04-12:
Great! Thank you!