w3c-cg/web-nfc #620: Invalid Web IDL loop between NDEFRecordInit and NDEFMessageInit

rakuco, 2021-08-31:
https://heycam.github.io/webidl/#idl-dictionaries says:

> The type of a dictionary member must not include the dictionary it appears on. A type includes a dictionary D if at least one of the following is true:
> 1. the type is D
> 1. the type is a dictionary that inherits from D
> 1. the type is a nullable type whose inner type includes D
> 1. the type is a sequence type or frozen array whose element type includes D
> 1. the type is a union type, one of whose member types includes D
> 1. the type is a dictionary, one of whose members or inherited members has a type that includes D
> 1. the type is record<K, V> where V includes D

#454 introduced a case of 4, 5 and 6, as right now `NDEFRecordInit.data` can be an `NDEFMessageInit`, which is a dictionary with a sequence of `NDEFRecordInit`s. This creates a loop that is not allowed by Web IDL and which can cause implementation issues like https://bugs.chromium.org/p/chromium/issues/detail?id=1242274

kenchris, 2021-09-08:
External records allow sub-records, so it does make sense that you can create a record with other records inside it. NFC doesn't define any limit of sub records with sub records, but doing more than one or two would be quite rare.

zolkis, 2021-09-09:
If we need to break the structural definition of recursive data, we can define a _function_ in NDEFRecord to init its `data` from an NDEFMessageInit. A breaking change :).

kenchris, 2021-09-09:
Maybe @littledan or @domenic have ideas here

reillyeon, 2021-09-09:
Prior to #454 the `data` field used the `any` type instead. This violates the spirit of the WebIDL specification but would give us the opportunity to add a depth limit to the [create NDEF message](https://w3c.github.io/web-nfc/#dfn-create-ndef-message) steps because we get to define when and how that arbitrary value is interpreted as an `NDEFMessageInit`.

domenic, 2021-09-13:
Yes, using `any` or `object` and doing the conversion in prose with an additional depth-check step seems like a good solution.

beaufortfrancois, 2021-09-14:
I'll work on a spec PR in the following days.

beaufortfrancois, 2021-09-15:
I've started a spec PR at https://github.com/w3c/web-nfc/pull/621. Please let me know what you think.

beaufortfrancois, 2021-09-17:
@rakuco Spec has been updated.

closed 2021-09-17 by beaufortfrancois