w3c-cg/web-nfc #622: Define NDEFMessage's and NDEFRecord's constructors normatively

rakuco, 2021-09-17:
We have the following in Web IDL:

```webidl
interface NDEFMessage {
  constructor(NDEFMessageInit messageInit);
  // ...
};

interface NDEFRecord {
  constructor(NDEFRecordInit recordInit);
  // ...
};
```

There is nothing in the spec saying what `NDEFMessage`'s constructor is actually supposed to do and how it is supposed to use `messageInit`. I can see _some_ explanation about `NDEFRecord` and `NDEFRecordInit` [here](https://w3c.github.io/web-nfc/#the-ndefrecord-interface), but some of the description of what to do with some of `recordInit`'s members is in this section and the rest spread across the spec, which makes it quite confusing to follow.

rakuco, 2021-09-17:
@beaufortfrancois this is also related to #620, as at the moment the only entry point for setting a recursion depth is `NDEFReader.write()`, but it's also possible to manually create `NDEFMessage` and `NDEFRecord` objects.

beaufortfrancois, 2021-09-22:
@zolkis Do you have time to look at it?

zolkis, 2021-09-22:
> @zolkis Do you have time to look at it?

Not right now, unfortunately - and not until the end of next week.

beaufortfrancois, 2021-10-04:
@zolkis gentle ping

zolkis, 2021-10-18:
IIUC, specifying a constructor for NDEFReadingEvent is not really necessary, as the default steps for ["firing an event"](https://dom.spec.whatwg.org/#concept-event-fire) apply and the rest (serialNumber and message) is specified in the [dispatch](https://w3c.github.io/web-nfc/#dispatching-nfc-content) step.
I made a section for that constructor, but feels trivial and obsolete. Anyway, if it helps clarity, I will include it.

rakuco, 2021-11-05:
* NDEFReadingEvent
  * https://dom.spec.whatwg.org/#concept-event-create and https://dom.spec.whatwg.org/#inner-event-creation-steps cover it. The latter has the algorithm to convert an `init`-like dictionary to members. We are implicitly relying on the `NDEFMessageInit` -> `NDEFMessage` conversion, which I'm not sure is very common. I wonder if you could just take an `NDEFMessage` directly without making a backwards-incompatible change?
* NDEFReader
  * While the constructor looks trivial, Web IDL [added](https://github.com/whatwg/webidl/pull/882) language a while ago that simplifies specifying constructor steps, and it does say all constructors need to be specified: https://webidl.spec.whatwg.org/#idl-constructors. See https://dom.spec.whatwg.org/#dom-eventtarget-eventtarget for example. My guess is that saying the constructor must do nothing + specifying the initial values for the internal slots should be fine.
* NDEFRecord/NDEFMessage
  * I don't think the constructor is trivial, as it needs to process `recordInit` somehow, and that includes hooking into "Creating NDEF message" like you said. All I could find so far was https://w3c.github.io/web-nfc/#dom-ndefrecordinit saying _"The NDEFRecordInit dictionary is used to initialize an NDEF record with its record type recordType, and optional record identifier id and payload data data"_. There is nothing saying how this initialization is supposed to take place though. Specifically, if I write `new NDEFRecord(myRecordInit)` there is nothing in the spec saying what must happen (including how to initialize `recordsDepth`).
  * The same applies to `NDEFMessage` and `NDEFMessageInit` in https://w3c.github.io/web-nfc/#the-ndefmessage-interface -- the constructor needs to perform non-trivial steps, and the text just says "The NDEFMessageInit dictionary is used to initialize an NDEF message" without saying how such initialization is supposed to happen (this isn't derived from DOM's `Event` interface, so you need to outline all the initalization steps).
  * With that said, section 9.10 ("Creating NDEF message") has an algorithmic descriptions of both [Create NDEF record](https://w3c.github.io/web-nfc/#creating-ndef-record) and [Creating NDEF Message](https://w3c.github.io/web-nfc/#creating-ndef-message) that pretty much correspond to the construction steps that are missing above. You could try doing something like this for NDEFRecord, for example:
    * Rename the "create NDEF record given record, context, and recordsDepth" operation to "initialize NDEF record given record, recordInit, context and recordsDepth". `record` would be an `NDEFRecord` object that you take as an argument rather than creating here (and the existing `record` argument becomes `recordInit`, which is also clearer).
    * Define e.g. `NDEFRecord`'s construction steps as _"The new NDEFRecord(recordInit) constructor steps are to perform "initialize NDEF record" with this, recordInit, ??? [dunno what the context would be] and 0".
    * Other algorithms that invoke "create NDEF record" directly could be changed to something like "1) Let foo be a [=new=] NDEFRecord 2) Invoke "initialize NDEF record" with foo, recordInit, context, and recordsDepth".
  * Something similar would work for NDEFMessage, but it would also be good to specify what "add NDEFRecord to NDEFMessage" means in [Add ndef to output](https://w3c.github.io/web-nfc/#dfn-create-ndef-message).

zolkis, 2021-11-16:
> Something similar would work for NDEFMessage, but it would also be good to specify what "add NDEFRecord to NDEFMessage" means in Add ndef to output.

Note that `output` (internal binary representation to be used by lower layers) is not the same as `NDEFMessage` (which is a developer input to generate `output`), i.e. we don't make the promise to update back `NDEFMessage` as `output` is being generated. I think for the methods of the spec it should be enough the specify how to generate `output`, rather than exacting the steps to make `output` presentable as `NDEFMessage`, as this was not an intent and it's quite a mine field. If that is needed, let's handle it separately, since it would require revamping a lot of algorithms. Needs more discussion.

zolkis, 2021-11-16:
There we have the problem that `NDEFReadingEventInit` contains an `NDEFMessageInit`, which contains a sequence (values), as opposed to `NDEFMessage` which contains a `FrozenArray` (references). I am not sure if the conversion steps are standard (citation possible?), or need to be specified in each spec. I wonder if we can have `FrozenArray` in an event (i.e. `NDEFReadingEventInit` would contain `NDEFMessage` instead of `NDEFMessageInit`).