w3c-cg/web-nfc #623: Inconsistency on failure to parse URLs and absolute URLs

rakuco, 2021-09-21:
https://w3c.github.io/web-nfc/#mapping-url-to-ndef says:
> * Let url be the result of parsing record's data.
> * If url is failure, throw a TypeError and abort these steps.

https://w3c.github.io/web-nfc/#mapping-absolute-url-to-ndef says:
> * If the result of parsing record's data is failure, throw a SyntaxError and abort these steps.

They should both throw the same type of error.

beaufortfrancois, 2021-09-22:
You're absolutely right @rakuco. Thanks for catching!
Chromium implementation throws TypeError for both url and absolute-url records.

```cpp

// Create a 'url' record or an 'absolute-url' record.
static NDEFRecord* CreateUrlRecord(const String& id,
                                   const NDEFRecordInit& record,
                                   ExceptionState& exception_state) {
  // https://w3c.github.io/web-nfc/#mapping-url-to-ndef
  if (
      !record.hasData() || !record.data()->IsString()
  ) {
    exception_state.ThrowTypeError(
        "The data for url NDEFRecord must be a String.");
    return nullptr;
  }

  // No need to check mediaType according to the spec.
  const String& url = record.data()->GetAsString();
  if (!KURL(NullURL(), url).IsValid()) {
    exception_state.ThrowDOMException(DOMExceptionCode::kSyntaxError,
                                      "Cannot parse data for url record.");
    return nullptr;
  }

  return MakeGarbageCollected<NDEFRecord>(
      device::mojom::blink::NDEFRecordTypeCategory::kStandardized,
      record.recordType(), id, GetUTF8DataFromString(url));
}
```
See https://source.chromium.org/chromium/chromium/src/+/main:third_party/blink/renderer/modules/nfc/ndef_record.cc;l=219;drc=3176b81a49dc3811a2b55eb70df6499400d53470?q=absolute-url%20f:nfc%20f:cc$&ss=chromium

I've submitted https://github.com/w3c/web-nfc/pull/624 to address this spec issue.

rakuco, 2021-09-22:
> Chromium implementation throws TypeError for both url and absolute-url records.

Actually according to the code snippet above the Chromium implementation throws a SyntaxError for both URLs and absolute URLs (it does throw a TypeError, but for the previous spec steps).

beaufortfrancois, 2021-09-22:
/me needs to wake up... I'll fix #624.

closed 2021-09-23 by beaufortfrancois