w3c-cg/web-nfc #642: Is the external type validate part correct?

kenchris, 2022-06-21:
[image]

asciiDomain here is the NSS part of https://datatracker.ietf.org/doc/html/rfc2141.html#section-2.2

For instance there is the rule that if there is a % it must be followed by two hex numbers, that is not checked in the algorithm.

Apart from that  "/" | "?" | "#" are allowed but reserved so we should probably ignore those for now (which we are doing).

kenchris, 2022-06-21:
btw "_" is supported so why are we not allowing that?

[image]

kenchris, 2022-06-21:
Am I missing something here @zolkis @beaufortfrancois

beaufortfrancois, 2022-06-27:
FYI Here's the [current implementation](https://source.chromium.org/chromium/chromium/src/+/main:third_party/blink/renderer/modules/nfc/ndef_record.cc;l=91;drc=f0ec23f77f2076e7ef2042c129d162e59ba755f8) of this algorithm in Chromium:

```cpp

// https://w3c.github.io/web-nfc/#dfn-validate-external-type
// Validates |input| as an external type.
bool IsValidExternalType(const String& input) {
  static const String kOtherCharsForCustomType(":!()+,-=@;$_*'.");

  // Ensure |input| is an ASCII string.
  if (!input.ContainsOnlyASCIIOrEmpty())
    return false;

  // As all characters in |input| is ASCII, limiting its length within 255 just
  // limits the length of its utf-8 encoded bytes we finally write into the
  // record payload.
  if (input.IsEmpty() || input.length() > 255)
    return false;

  // Finds the first occurrence of ':'.
  wtf_size_t colon_index = input.find(':');
  if (colon_index == kNotFound)
    return false;

  // Validates the domain (the part before ':').
  String domain = input.Left(colon_index);
  if (domain.IsEmpty())
    return false;
  // TODO(https://crbug.com/520391): Validate |domain|.

  // Validates the type (the part after ':').
  String type = input.Substring(colon_index + 1);
  if (type.IsEmpty())
    return false;
  for (wtf_size_t i = 0; i < type.length(); i++) {
    if (!IsASCIIAlphanumeric(type[i]) &&
        !kOtherCharsForCustomType.Contains(type[i])) {
      return false;
    }
  }

  return true;
}
```

@kenchris Which changes are you suggesting?

kenchris, 2022-06-27:
For instance: "The presence of an "%" character in an URN MUST be followed by two characters from the <hex> character set."

kenchris, 2022-06-27:
So we check whether the string is an ascii string, meaning: An ASCII code point is a [code point](https://infra.spec.whatwg.org/#code-point) in the range U+0000 NULL to U+007F DELETE, inclusive.

[image]

Then we return false if it contains the following chars, plus underscore:: A forbidden host code point is U+0000 NULL, U+0009 TAB, U+000A LF, U+000D CR, U+0020 SPACE, U+0023 (#), U+002F (/), U+003A (:), U+003C (<), U+003E (>), U+003F (?), U+0040 (@), U+005B ([), U+005C (\), U+005D (]), U+005E (^), or U+007C (|).

This doesn't seem to be good enough to match the following:

0-9 | A-Z | a-z | "(" | ")" | "+" | "," | "-" | "." |":" | "=" | "@" | ";" | "$" | "_" | "!" | "*" | "'" |  '%" | "/" | "?" | "#"

Even _ seems to be allowed there. And there is a special restriction on % as it has to be followed with two hex values

kenchris, 2022-06-27:
It just looks like the ascii domain part is all wrong and I see that the implementation doesn't even check that: https://crbug.com/520391

kenchris, 2022-06-27:
There seems to be multiple definitions of NSS :-( https://datatracker.ietf.org/doc/html/rfc8141#section-2.2 : The NSS as specified in this document allows several characters not
   permitted by earlier specifications (see [Appendix B](https://datatracker.ietf.org/doc/html/rfc8141#appendix-B)).

kenchris, 2022-06-27:
So the newer specs says its:

NSS           = pchar *(pchar / "/")

pchar         = unreserved / pct-encoded / sub-delims / ":" / "@"
unreserved    = ALPHA / DIGIT / "-" / "." / "_" / "~"
pct-encoded = "%" HEXDIG HEXDIG
sub-delims  = "!" / "$" / "&" / "'" / "(" / ")" / "*" / "+" / "," / ";" / "="

It's basically the same as allowed for type + @ and &, as well as / (but not for first char) and the hex restriction for %

kenchris, 2022-06-27:
@zolkis do you remember why this is specced like it is? the hex validation definitely seems missing and we should probably not call it domain but nss

kenchris, 2022-06-27:
Seems like this should probably be:

> The external type nns percent-encode set is the [C0 control percent-encode set](https://url.spec.whatwg.org/#c0-control-percent-encode-set) and U+0020 SPACE, U+0023 (#), U+003A (:), U+003C (<), U+003E (>), U+003F (?), U+005B ([), U+005C (\), U+005D (]), U+005E (^), U+0060 (`), U+007B ({), U+007C (|), and U+007D (}).

> If input is empty, return false.
> [UTF-8 percent-encode](https://url.spec.whatwg.org/#string-utf-8-percent-encode) input using the the [external type percent-encode set].
> If input length exceeds 255 bytes, return false.
> Let nns be the input from the start of input up to but excluding the last occurrence of U+003A (:), or null if that is not found.
> Let type be the input after the last occurrence, if any, of U+003A (:) up to the end of input, or null if that is not found.
> If nns or type is null, return false.
> If nss starts with U+002F (/), return false.
> If nss contains [C0 control](https://infra.spec.whatwg.org/#c0-control), U+0020 SPACE, U+0023 (#), U+003A (:), U+003C (<), U+003E (>), U+003F (?), U+005B ([), U+005C (\), U+005D (]), U+005E (^), U+0060 (`), U+007B ({), U+007C (|), U+007D (})  or U+007F DELETE, return false
> If type contains [code points](https://infra.spec.whatwg.org/#code-point) that are not [ASCII alphanumeric](https://infra.spec.whatwg.org/#ascii-alphanumeric), or U+0024 ($), U+0027 ('), U+0028 LEFT PARENTHESIS ((), U+0029 RIGHT PARENTHESIS ()), U+002A (*), U+002B (+), U+002C (,), U+002D (-), U+002E (.), U+003B (;), U+003D (=), U+0040 (@), U+005F (_), return false.
> Return true.

beaufortfrancois, 2022-06-28:
@kenchris Can you start a spec PR?
I'll make sure Chromium implementation is conformant.

closed 2022-06-29 by kenchris

beaufortfrancois, 2022-06-29:
@kenchris Can you file a crbug at crbug.com/new?

kenchris, 2022-06-30:
Sure, https://bugs.chromium.org/p/chromium/issues/detail?id=1340827