w3c-cg/web-nfc #653: Read record from RFID card string UUID via nfc scan reader

ankitraut05, 2023-02-19:
[#WebNFC](https://mobile.twitter.com/hashtag/WebNFC?src=hashtag_click) I am using Web NFC in my project. i coming difficulties to read values from RFID card through scan card on nfc reader in Web application. can you help me for how to read hexadecimal, decimal (UUID) from RFID card.

beaufortfrancois, 2023-02-20:
Can you share the model of the NFC device and some of your code so that we can help you?
Do you get at least one NDEF record when you try https://googlechrome.github.io/samples/web-nfc/?

ankitraut05, 2023-02-20:
Hi,

I am using Sunmi V2 pro device.
I created a small html page and wrote javascript code. and i scan rfid card
through NFC device.
Can you help me to read the hexadecimal, decimal (UUID) from the RFID card?

Thanks,
Ankit Raut.

beaufortfrancois, 2023-02-20:
Can you share a screenshot of what kind of results you get?

ankitraut05, 2023-02-20:
PFA.

beaufortfrancois, 2023-02-20:
You may want to use GitHub website interface to upload the screenshot.

[image]

ankitraut05, 2023-02-20:
[image]

Open this link and see

beaufortfrancois, 2023-02-20:
Thank you! Now, where's the value you're trying to read? It looks like you only have one NDED record for now.
If you wanted to write an hexadecimal value for instance, you could do something like below:
```js
const ndef = new NDEFReader();
await ndef.write(0x1234);
```
And for reading it, you could do:
```js
const textDecoder = new TextDecoder(record.encoding);
const text = textDecoder.decode(record.data);
console.log(text.toString(16));  // 1234
}
```
FYI https://developer.chrome.com/articles/nfc/#read-and-write-a-text-record contains plenty of code examples.

ankitraut05, 2023-02-20:
When i write data to card the i getting message "Argh! Network Error :
Failed to write due to an IO error : null"

Thanks,
Ankit.

beaufortfrancois, 2023-02-20:
Then it means you can't write to this device right? Can you copy/paste the result of NDEF records you successfully read?

ankitraut05, 2023-03-12:
Hi,

NDEF Record successfully read on device and running properly.

Can i running NDEF record without https.means can i run on http.

eklem, 2023-03-12:
[It's only available in secure context](https://developer.mozilla.org/en-US/docs/Web/API/NDEFReader), @ankitraut05
I'm guessing every production environment should be https, but it makes it a bit more difficult when developing.

I'm using VSCode and using Ngrok and Live Server to make it work when developing. Just a little work the first time, and then it works smoothly. [Here's a little bit more about the setup I use](https://github.com/eklem/nfc-json-transfer#development-setup).