w3c-cg/web-nfc #654: Read NFC tag after writing to it

beaufortfrancois, 2023-04-04:
It is currently cumbersome to read records from an NFC tag after writing to it as the user must tap the NFC tag one first time to write to it, then make it out of range so that they can tap it again to read records.

To help with that journey, I was thinking we could perform a read NFC operation after a write NFC operation and return the `NDEFMessage` when the `NDEFReader.write()` promise resolves if web developers expressed it via an `NDEFWriteOptions` option.

```js
// Waiting for user to tap NFC tag...
const ndef = new NDEFReader();
await ndef.scan();

// Write and read NFC tag.
const message = await ndef.write("Hello world", { readAfter: true }); // <-- NEW!
console.log(message);
```

Note that by setting the `readAfter` default value to `false`, the current behavior does not change for existing web apps using `NDEFReader.write()` today.  

```webidl
interface NDEFReader : EventTarget {
  ...
  Promise<NDEFMessage?> write(NDEFMessageSource message,
                              optional NDEFWriteOptions options={});
};

partial dictionary NDEFWriteOptions {
  boolean readAfter = false;
};
```

kenchris, 2023-04-04:
I am supportive of this

zolkis, 2023-04-04:
+1 (why not)

Remind me, what was the use case? 
Convenience for single tap, when wanting to make sure the written content is the same as the one at hand when writing?

closed 2023-04-26 by beaufortfrancois

kenchris, 2023-04-27:
Maybe it merits an example in the spec or in your article then?

zolkis, 2023-04-27:
> This is actually not needed and can be exercised with the JavaScript code below. Sorry for the noise.
> // Waiting for user to tap NFC tag to write to it...
> // Now scanning for 3 seconds...
> // Scanning is now stopped.

So from the user perspective, the tag is just being held there for 3 seconds (or less, until read), and that solves the original issue, that a user was assumed to need to tap, lift, and tap again?

But would the second part still make sense for convenience? IIRC we used to have this: what was the reason for removing? Occam, or something more?

> I was thinking we could perform a read NFC operation after a write NFC operation and return the NDEFMessage when the NDEFReader.write() promise resolves if web developers expressed it via an NDEFWriteOptions option.