w3c-cg/web-nfc #659: Write to password protected NFC

ShriramKhandbahale, 2024-02-16:
In the Android NFC Tools app, there is an option to set a password for the NFC to make it write-protected, and next time I try to write to that NFC it gives a write error, so If I want to update it, I have to remove the password by providing it my old password and then write the data to it and then again protect it with the password.

but now when I try to write it with web nfc api, It gives me the write error, is there a way I can write to a write-protected nfc through web nfc api by providing it the password that I set from NFC tools app?

beaufortfrancois, 2024-02-16:
I'm not familiar with how the NFC Tools app works. It would help us understand how/if we can make Web NFC works better if we knew how write-protected nfc tags with this app.

dtinth, 2024-04-10:
The NFC tools app has these 2 menu items:

- Set password
- Remove password

[image]

The **Set password**, when pressed, asks for a password. Then it makes the NFC tag read-only (writes will fail). It is similar to locking the NFC tag with `.makeReadOnly()` but it is temporary. If you can provide the same password to the **Remove password** function, the unlocks the NFC, making it read-write.

Therefore, I'd suggest the following API (name should be changed but you get the idea):

```js
NDEFReader.makeReadOnlyWithUnlockPassword('the-unlock-password') // => Promise

NDEFReader.unlockWithPassword('the-unlock-password') // => Promise
```

dtinth, 2024-04-11:
From my additional research, it seems that this Password feature is not NFC Forum standard, but is an additional feature by some NFC manufacturers. For example, [here’s the C code for setting up password in NTAG213/215/216 tags](https://github.com/nfc-tools/libfreefare/blob/c2b0cfa4b9fb0e4be88604f00b7a2405618d5abc/libfreefare/ntag21x.c#L357), and the [related data sheet](https://www.nxp.com/docs/en/data-sheet/NTAG213_215_216.pdf).

See also, https://stackoverflow.com/a/36592599/559913

wakdev, 2024-06-28:
The password feature is not a NFC Forum standard but a NTAG21x feature. You can check the datasheet to see how it's works in detail. In NFC Tools, we use the first bytes of the MD5 hash of your password. Because NTAG chips requires a 4 bytes password. For non-tech-savvy users, we decided to ask for a simple text password instead of four Hex values (or more depending of the chips type).

wakdev, 2024-06-28:
The password feature is not a NFC Forum standard but a chip related feature. You can check the datasheet to see how it's works in detail. In NFC Tools, we use the first bytes of the MD5 hash of your password. Because, as an example, NTAG chips requires a 4 bytes password. For non-tech-savvy users, we decided to ask for a simple text password instead of four Hex values (or more depending of the chips type).

beaufortfrancois, 2024-07-01:
Thank you @wakdev! 

Since it's not a NFC Forum Standard, I believe we won't be implementing this in Web NFC.