w3c-cg/web-nfc #664: Broken link for "Latest published version"

nondebug, 2024-09-04:
The "Latest published version" link in the spec header links to https://www.w3.org/web-nfc/ which does not exist.

beaufortfrancois, 2024-09-05:
Thanks for raising this issue @nondebug!

@anssiko Is there something missing in the respec config there that would cause https://www.w3.org/web-nfc/ to return 404? I'd prefer us fixing the root issue rather than suppressing the  "Latest published version" link if we can.

anssiko, 2024-09-05:
@beaufortfrancois this bug was discussed in https://github.com/speced/respec/issues/1562 and the "none" fix was introduced in https://github.com/speced/respec/pull/2968

A one line fix to this bug is to use the "respec-w3c-common" profile instead of the "respec-w3c" profile. See the first `<script>` in the document.

The "respec-w3c-common" profile renders this correctly:

>Latest editor's draft:
>https://w3c.github.io/web-nfc/

While the "respec-w3c" profile outputs a broken link:

>Latest published version:
>https://www.w3.org/web-nfc/

@tidoust may still remember if there's a fix available using the "respec-w3c" profile for CG Draft Reports. He's hacked with this tool quite a bit :-)

beaufortfrancois, 2024-09-05:
Thank you @anssiko! I've just tried your suggestion with https://github.com/w3c/web-nfc/pull/666 and it seems to work great. Feel free to review it

beaufortfrancois, 2024-09-05:
#666 has been merged. Thanks both of you!

closed 2024-09-05 by beaufortfrancois

tidoust, 2024-09-05:
> @tidoust may still remember if there's a fix available using the "respec-w3c" profile for CG Draft Reports. He's hacked with this tool quite a bit :-)

Good. FWIW, I had forgotten about the fix, thanks for the reminder! (I didn't even remember having raised the issue in Respec in the first place ;))